Ultimaker/Cura #12640: Initial Layer Wall Speed

weaseljohnson, 2022-06-29:
### Is your feature request related to a problem?

A lot of models can have complex initial layers. The complexity requires that the initial layer be printed much slower than the rest of the model. With those complex models, I have to run between 10 mm/s to 15 mm/s most of the time. However, once the outer walls are finished printing, the rest of the layer takes ages to fill in due to the slow speeds. 

### Describe the solution you'd like

If there was a setting for just the initial layer wall speed, that would solve the problem and drastically improve print times for a lot of people. 

### Describe alternatives you've considered

I'm unaware of any alternative solutions to this issue.

### Affected users and/or printers

This feature could benefit all users of Cura and all FDM printers.

Ghostkeeper, 2022-07-13:
We've decided to defer this, for a combination of reasons:

* The initial layer, being only one layer, is not a big time saver in the grand scheme of things.
* We have a lot of settings for the first layer specifically and we're getting to the point where we really need a different system for it, lest we end up basically duplicating all of our settings for the first layer.
* In our experience, the skin itself is also prone to failure during the first layer, and needs to be printed at a similar speed to the walls. The primary cause of this is that the first layer has a greater layer thickness and if there is any underextrusion, the force of the extrusion is not good enough to make it stick. If there is any overextrusion, the nozzle tends to pull the print off the build plate. So while you might theorise that it may improve productivity, we think that this will not provide a good ratio of productivity to reliability.

MIOsystems, 2022-07-14:
>     * We have a lot of settings for the first layer specifically and we're getting to the point where we really need a different system for it, lest we end up basically duplicating all of our settings for the first layer.

This would be a great idea. I came to this issue because i had the same idea. Because me and definitely many other (PRO) printnerds will benefit from it. "Print Speed - Initial Layer - Outer Wall" would be the solution to get the first crucial lines right. Infill is never a problem because these are straight lines but they spend most of the time

obertini78, 2022-11-01:
I agree with your first point. The first layer is mostly not the main driver for the total print time. 
However, I share the same opinion as  MIOsystems. I tend to wait until the first layer is fully printed until I can get away from the printer. I'm sure I'm not the only one. Having to wait for the first layer infill to be completed at 10-20 mm/s is a time killer.
Up to you guys of course to reconsider or not this feature request. Keep the great work going!

smartavionics, 2022-11-01:
See https://github.com/smartavionics/Cura/releases for a Cura variant that has supported setting speed/accel/jerk for the initial layer's walls since 2017!

javawizard, 2023-03-19:
If someone were to go to the effort to code this up, is there any likelihood a PR would be accepted?

For my use case, the reasons to defer in https://github.com/Ultimaker/Cura/issues/12640#issuecomment-1183014705 don't make much sense:

> The initial layer, being only one layer, is not a big time saver in the grand scheme of things.

I print a lot of flat, wide objects - think trays or the bottom of a drawer. The most recent one took just under an hour for the initial layer followed by 1 hour 40 minutes for the remaining 15 layers. Being able to print the initial layer skin at 400% speed would represent a ~30% reduction in print time which is significant.

> In our experience, the skin itself is also prone to failure during the first layer, and needs to be printed at a similar speed to the walls. The primary cause of this is that the first layer has a [...]

There's one trick that - in my extensive **experimentation** (i.e this is not just theoretical) - resolves this almost entirely: setting a low initial layer acceleration speed. With an initial acceleration of 300 mm/s², an initial layer wall speed of 25 mm/s, and an "initial layer skin speed" (manually simulated by setting the print speed to 400% while the skin is printing) of 100 mm/s, the skin prints perfectly and indistinguishably from a much slower speed.

I'm going to be printing a run of ~120 of these flat, wide objects in the next month and the speed improvement would be worth enough that I would seriously consider writing up a PR to implement this if there's any likelihood it would be accepted. If that's not the case, I will content myself with hacking the gcode to work around this for now.

ThVandy, 2023-10-26:
I would also like to see this pushed through. I don't want to have to get a custom cooked up version of Cuda just to be able to control the initial wall speed. I also print flat wide bodies that have very intricate edges that unless I'm printing 10mm/s they won't stay in place. After the 3 walls are down, it takes an hour and 45 minutes to print the rest of the first layer at 10mm/s. Just stating I would like to see it added or the option for it is all. Thanks again for all the great work you guys have done!

TamaBaka, 2023-11-04:
I found myself needing it because I tend to have a lot of failures on the first layer due to bouncing back and forth between different kinds of filaments, each with their own sticking behavior and settings: PLA, PETG, ASA.  Once I get past that, then I'm good to go, but the moment blobs form or edges rip right off the print bed, I have to start all over again.  

3 failures, where the first layer takes 15 minutes at a time, is not something to take lightly.   Getting the first layer to lay down quickly so I can fail fast and fix what I forgot would make me happy.

TjeerdvanDijk, 2024-01-25:
I usually never comment but I have been printing with cura for 6 years and almost every single print this has been bothering me. I print large objects and usually i just sit there untill all the walls are finished before i crank the speed up to 500%, then i have to wait for it to finish to turn the speed back to 100%.

Can we please PLEASE make an exception and add this SINGLE extra feature for the first layer. Maybe in the experimental category ?

obertini78, 2025-01-25:
Before this request is closed, here is a workaround:
One can use the [astFirstInfill.py](https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/FastFirstInfill.py) from https://github.com/5axes/Cura-Postprocessing-Scripts.

Set a low speed for the first layer and use the script to increase the first layer infill speed.