Ultimaker/Cura #12641: Unable to slice with the current settings. The following settings have errors: Z Hop Speed.

dperetti, 2022-06-30:
### Application Version

5.0.0

### Platform

Mac OS

### Printer

Alfawise u30

### Reproduction steps

Brand new Cura 5.0 install on brand new mac.
Alfawise U30 printer.
Load simple stl file from Fusion.

### Actual results

Unable to slice with the current settings. The following settings have errors: Z Hop Speed.
[image]

### Expected results

Should have sliced.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[STL_and_3MF.zip](https://github.com/Ultimaker/Cura/files/9017552/STL_and_3MF.zip)

GregValiant, 2022-06-30:
Thanks for the report.
In your settings you have the Z-hop speed set to "10mm/sec" but your printer definition file lists the max Z speed as "5mm/sec" and so it won't slice.
In the Cura Speed settings, set your Z-Hop speed to 5 and it will slice.
This is the line from your printer definition file that is causing your problem.
        "machine_max_feedrate_z": { "default_value": 5 },
In the Cura MarketPlace is a plugin called "Printer Settings".  If you load it you can access the basic printer settings.  It will have a line where you can change the "Maximum Speed Z".
The printer definition file may have been written with that limit because the M203 setting in the printer might have the Max Speed Z at 5.

nallath, 2022-07-05:
I've ensured that the default state isn't in an error mode anymore.

closed 2022-07-05 by nallath