Ultimaker/Cura #12642: Faster Travel Speed lead to longer print

trivalik, 2022-06-30:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Anet A6

### Reproduction steps

Load a big model, enable expert settings, slice note the required time then increase travel speed to i.e. 200, then the required print time increases.

### Actual results

The print time increases instead of decreases.

### Expected results

The print time should decrease.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

GregValiant, 2022-06-30:
Thanks for the report.
Print times are extremely model and printer specific.  I can't duplicate what you are describing on my computer.
Use the "File | Save Project" command to create a 3mf project file.  Zip the file and post the Zip file here.

trivalik, 2022-06-30:
Please find attached one with 175 and one with 200 travel speed. With 200 is it one minute less. I was using last time another infill then the value difference was bigger.
[example.zip](https://github.com/Ultimaker/Cura/files/9019944/example.zip)

GregValiant, 2022-06-30:
Thanks for the report.
There are a lot of settings that come into play when calculating Print Time.  The main setting is not Travel Speed, but Acceleration and with the acceleration a constant, it takes longer to accelerate from 0 to 200 than it does from 0 to 175.
On that model there are a lot of combing moves but there are a lot more very short travel moves in the skins.
Notice that if you change the Top Bottom pattern from Lines to ZigZag the print time drops by 18 minutes or about 9%.
To the right of the "Print Time" is an "i" in a circle.  When you hover the mouse over it the breakdown of the print time will appear.
Someone from the Cura team will take a look at this.

Joeydelarago, 2022-07-06:
Hey, I've investigated this issue and found that increasing the travel acceleration also barely changes the print time. This could possibly be an issue with the time estimation calculations. Thanks for bringing this to our attention, I'll discuss with the team.

Ghostkeeper, 2022-07-13:
Developers, see issue CURA-9474.