Ultimaker/Cura #12643: Cura 5.0.0 closed unexpected and crashes on every start

chiefmaster81, 2022-06-30:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

1. Cura closed unexpected
2. since then cura crashes on every start.
3. reboot did not help
4. reinstall did not help
5. deleteing files and folders and reinstall did noct help

### Actual results

cura crashes on every start.

### Expected results

cura opens and works properly

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/9019375/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/9019376/cura.log)

Joeydelarago, 2022-07-06:
Hello, are you using a desktop plugged into a monitor?  Also if you plug your monitor into your motherboards hdmi slot does this crash still happen?

Based on the logs I think this could be a duplicate of #12602

inocuosistemas, 2022-07-27:
Same problem here...also with Cura 5.1, every start after Loading UI...shows the Cura window in white color, and closes