Ultimaker/Cura #12645: Slicing Failure of object with 5 components

inspectionsbybob, 2022-06-30:
### Application Version

5.0.0 

### Platform

Win10

### Printer

Custom JGMaker Artist-Dthat

### Reproduction steps

Tried to slice the enclosed EyeBox that had 6 pieces

### Actual results

Slice failure with no useful error message

### Expected results

Completed slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[EyeBox.zip](https://github.com/Ultimaker/Cura/files/9020504/EyeBox.zip)

No Logfile found at  C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

GregValiant, 2022-06-30:
Thanks for the report.
Did the warning come up that "The model is not watertight and may not print correctly"?  The model with the external threads seems to be the problem.  You can repair it using MS 3D Builder, Mesh Mixer, or another repair utility.
When I open your 3mf file I don't see your printer or settings profile (all I got are the models).  Did you use the "File | Save Project" command to create the file?
With a Creality CR-10 as the active printer with the Standard profile the models sliced.  At this point I am unable to duplicate this.

GregValiant, 2022-07-05:
I had no problem slicing this with 5.1 beta.  It appears that this bug has been fixed.
I'll close this.  If you have further comments you can still post them here and I'll be notified.

closed 2022-07-05 by GregValiant