Ultimaker/Cura #12648: Cura 5 UFP files not processed on Klipper

MrEd88, 2022-06-30:
### Application Version

5.0.0

### Platform

Linux Mint 20.2

### Printer

Modified Ender 3

### Reproduction steps

1. Slice model and save as ufp
2. Try uploading it in Klipper ( mainsail and Fluidd)

### Actual results

File is stuck on proccessing and doesn't upload/convert to gcode.

### Expected results

This may be a klipper-monraker-fluidd-mainsail issue, however it is working fine with Cura 4.13 so something changed with Cura 5 ufp format. I am using ufp as this format allows klipper to show thumbnail images. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N.A.

fieldOfView, 2022-07-01:
This is definitely a "klipper-moonraker-fluidd-mainsail" issue. The UFP writer did not have functional changes between Cura 4.13 and 5.0.

Please collect logs on the klipper-moonraker-fluidd-mainsail side of this, and report the issue there.

closed 2022-07-01 by fieldOfView

Yosoy-Ed, 2024-11-29:
Please close this. Issue has not been observed again and was several versions ago. Sorry for not providing logs at the time.