Ultimaker/Cura #12649: slicing issue

poolboy97, 2022-07-01:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

cr10

### Reproduction steps

I have deleted file and reload, restarted cura multiple times, tried different models. Have checked for updates, none available.

### Actual results

Holes within the model load properly but are filled solid once sliced and printing.

### Expected results

Holes for screws to be holes and not solid surface

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[CCR10_Upper Main (1).zip](https://github.com/Ultimaker/Cura/files/9025079/CCR10_Upper.Main.1.zip)
[cura.log 4.13.zip](https://github.com/Ultimaker/Cura/files/9025135/cura.log.4.13.zip)

Asterchades, 2022-07-01:
As is most commonly the cause when this happens, you need to disable the Remove All Holes meshfix. It's doing what you've told it to do.

closed 2022-07-01 by Asterchades

poolboy97, 2022-07-01:
Can you direct me were to do that? I must have accidentally done then and not sure were. 

Larry Shirley

Asterchades, 2022-07-01:
You can use the search box to look for settings by name. That will even show them if they're hidden, which that setting appears to be in your project file. The exact setting name is "Remove All Holes".

Alternatively you can make the setting visible to access it directly. Click the menu "hamburger" next to the setting search box and select "Manage Setting Visibility...". From there use the search box again or manually look for "Remove All Holes" (it will be the second setting under Mesh Fixes) and check the box next to it.

In either case there will be an undo icon next to the setting, showing that it has been altered from the default at some point. I can't even begin to guess as to how or when (you would be in a better position to answer that) but either hitting that undo circle or unchecking the box will remedy the situation.

poolboy97, 2022-07-01:
I will check that out. Thank you so much!

Larry Shirley