Ultimaker/Cura #12650: Cura 5 slicing example like the old engine (4.13)

YardenNaveh, 2022-07-01:
### Application Version

5.0.0

### Platform

MacOs Big Sur

### Printer

Ender 3

### Reproduction steps

Uploaded my model and sliced it. 

### Actual results

The perimeter is printed with very tiny lines instead of long lines along the sides. 
My understanding is that this is exactly what the new engine should resolve, so I'm sending this as an example. 

### Expected results

Perimeter should be printed with long lines along the sides. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_Body1.3mf.zip](https://github.com/Ultimaker/Cura/files/9026597/CE3PRO_Body1.3mf.zip)
[image]

GregValiant, 2022-07-01:
I opened your project file in 5.0 and sliced it.  I didn't get that skin area.  You can see below that there are parallel lines there.  I'm not sure what to make of this as I can't duplicate it even using your project.
[image]

This is the project sliced in 4.13.1.  Looks the same as 5.0 with lines around the perimeter.  I sliced it in 4.8 as well and didn't get the problem.  
I don't know what information you could add, but if you can think of something pass it along. 
[image]

GregValiant, 2022-07-03:
In the image below you can see that Section B-B is a narrow ring and Section A-A is a lot wider.  The extra width of A-A would mean that the Cura Outer Walls on that layer are far apart and the area between those outer walls would be filled with skin (not infill) in order to maintain the wall thickness through that area.  It is the same thing that is happening with your model (though the image is on a larger scale and has larger radii).
[image]

I'm inclined to close this as it is "expected behavior" and a function of the geometry of your model coupled with the settings you are using.  Let me know what you think.

closed 2022-07-21 by github-actions[bot]