Ultimaker/Cura #12653: "Print as support" not working as expected?

EvillNooB, 2022-07-01:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Imported custom supports along with the model
2. Changed their printing mode to "print as support"
[image]

### Actual results

Shape gets replaced with ordinary support, so "print as support" works as a support blocker in reverse, marking the model parts to be supported, but not actually printing provided 3d model "as support"
[image]

### Expected results

Shapes should remain the same, but their settings should be replaced with support settings (infill, wall count, support interfaces and other unique settings) 
basically, same shape, but printed as if it was a support
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[test_part2.zip](https://github.com/Ultimaker/Cura/files/9031700/test_part2.zip)

GregValiant, 2022-07-01:
It doesn't like that wedge shape for some reason.  Someone from the Cura Team will take a look.

fieldOfView, 2022-07-03:
This is because of the setting "Drop Down Support Mesh", which defaults to true. In the per model settings tool, where you have set the wedges to "print as support", add the "Drop Down Support Mesh" and then turn it off. You are going to have to do that for each of the models.

closed 2022-07-03 by fieldOfView