Ultimaker/Cura #12659: Save to Disk does not automatically populate a filename anymore

JeffreyFalgout, 2022-07-02:
### Application Version

5.0.0

### Platform

Ubuntu 20.04

### Printer

Creality Ender-3 Pro

### Reproduction steps

 - Open an STL file into Cura
 - Slice it
 - Click Save to Disk

### Actual results

A save file window pops up with nothing pre-filled for the filename

### Expected results

I expect the filename to be prefilled as "CE3PRO_xyzCalibration_cube.gcode". 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura 4.13.1: https://user-images.githubusercontent.com/2907401/177017962-96707719-2a45-412b-a741-2b819315990f.mp4

Cura 5.0.0: https://user-images.githubusercontent.com/2907401/177017898-0b5c994e-a615-42a5-834e-4a63e5e2b2bf.mp4

JeffreyFalgout, 2022-07-02:
I think this is probably related to https://github.com/Ultimaker/Cura/issues/11991. If I insert a removable drive and click "Save to Removable Drive", the file is saved correctly as `CE3PRO_xyzCalibration_cube.gcode`

JeffreyFalgout, 2022-07-02:
Here's a log of me opening a fresh instance of Cura, opening a calibration cube, slicing it, and clicking save to disk (note, I didn't see any logs when I clicked save to disk). I've also included a project file for that instance (I renamed it to zip since github wasn't happy with .3mf)

[cura.log](https://github.com/Ultimaker/Cura/files/9033813/cura.log)
[cura_project.zip](https://github.com/Ultimaker/Cura/files/9033814/cura_project.zip)

JeffreyFalgout, 2022-07-02:
Looks like "Save to Disk" is actually implemented over in Uranium. Let me know if you'd rather me file a bug over there

https://github.com/Ultimaker/Uranium/blob/99f8ac11cfc21095a9d3776908234da95ff69c8c/plugins/LocalFileOutputDevice/LocalFileOutputDevice.py

rgomezjnr, 2022-07-03:
> I think this is probably related to #11991. If I insert a removable drive and click "Save to Removable Drive", the file is saved correctly as `CE3PRO_xyzCalibration_cube.gcode`

If you do this then does the file format in GFF+ work?

fieldOfView, 2022-07-03:
The problem seems to be in the communication between Cura and the system-provided file dialog; for some reason, on some Linuxes, the dialog ignores the filename that gets passed to it. The "Save to Removable Drive" skips the file dialog, and thusly gets the correct name.

GFF+ would not fix this.

fieldOfView, 2022-07-03:
Probably a duplicate of https://github.com/Ultimaker/Cura/issues/12295, though that also mentions some other stuff that may not be the same issue.

JeffreyFalgout, 2022-07-03:
> > I think this is probably related to #11991. If I insert a removable drive and click "Save to Removable Drive", the file is saved correctly as `CE3PRO_xyzCalibration_cube.gcode`
> 
> If you do this then does the file format in GFF+ work?

Yep. GFF+ does work when saving to a removable disk, so this is definitely a cura issue and not a GFF+ issue (https://github.com/rgomezjnr/GcodeFilenameFormatPlus/issues/11)

JeffreyFalgout, 2022-07-03:
I'm not too sure if this is specific to the file system dialog. When I run the following snippet on the same machine I'm using to rune cura, I get a file dialog box pre-populated with `foo`

```
#!/usr/bin/env python3

from PyQt6.QtWidgets import QApplication, QFileDialog

app = QApplication([])

d = QFileDialog()
d.setFileMode(QFileDialog.FileMode.AnyFile)
d.setAcceptMode(QFileDialog.AcceptMode.AcceptSave)
d.selectFile("foo")

d.exec()
```

What version of python / PyQt6 is Cura bundled with?

```
$ python3 --version
Python 3.8.10

$ pip show PyQt6
Name: PyQt6
Version: 6.3.1
Summary: Python bindings for the Qt cross platform application toolkit
Home-page: https://www.riverbankcomputing.com/software/pyqt/
Author: Riverbank Computing Limited
Author-email: info@riverbankcomputing.com
License: GPL v3
Location: /home/jeffrey/.local/lib/python3.8/site-packages
Requires: PyQt6-Qt6, PyQt6-sip
Required-by: 
```

JeffreyFalgout, 2022-07-03:
I get the same behavior with PyQt6 6.2.3 too (I think that's the version being used by cura based on https://github.com/Ultimaker/cura-build-environment/blob/a8a4b6b169ac8024fbdaf17f8f0b1e9f88521f24/projects/requirements.txt)

JoshMcCullough, 2023-02-12:
On Manjaro Linux, the filename is filled in in the save dialog but `.gcode` is not appended (have to type it manually).

MariMakes, 2023-03-31:
Quick update on our side. 

We might have a fix for this issue. 🎉 
It's a bit hard for us to test, but we have a nightly build available. 
Can you please try it out and let us know if it resolves your issue? 

[You can find the download links and the discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)

robertkardos, 2023-03-31:
Hi,
I've just tested this, and found the following:

It populates the filename field automatically when I try to save either the project or the gcode, though it seems to work differently from Cura 4.x where after saving the project file itself did not cause the gcode to take the new name automatically, but in this nightly after saving the project under a different name, the filename for the gcode was automatically the new one.

### So in Cura 4:

- ctrl S opens up **the current location** of the current project file with the current name of the project file, for example WHATEVER.3mf
- I save it as WHATEVER_cura4.3mf
- I click "Save to Disk" and it opens up the current location of the current gcode file with **the original name of the gcode file**, WHATEVER.gcode
- If I save either the project file or the gcode under a different name, next time it wants to save in under the original name WHATEVER.3mf or WHATEVER.gcode

I guess this based the name on the .stl file.

### In the Cura Nightly you have linked:

- ctrl S opens up **the home directory** but now with the current name of the project file, for example WHATEVER.3mf
- I save it as WHATEVER_curaNightly.3mf
- I click "Save to Disk" and it opens up the home directory but now with **the current name of the project file**, WHATEVER_curaNightly.gcode
- If I save the project file under a different name, the gcode file wants to inherit that new name, so if I save the project as WHATEVER_curaNightly_new.3mf and the gcode as WHATEVER_curaNightly_newX.gcode, the next time I try to save the gcode, it wants to save it with the current project name, so it will be WHATEVER_curaNightly_new.gcode.

I don't know if this new way is the intended behavior, I guess it has pros and cons.

robertkardos, 2023-03-31:
As it was mentioned in this comment https://github.com/Ultimaker/Cura/issues/12659#issuecomment-1173046996 this seems to be half of the issue they have linked.

The other half of that problem still stands though, which is the far more irritating one, but if this is solved, maybe it might help solve the other half?

MariMakes, 2023-08-28:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue regarding populating the filename. This seems to have been resolved in Cura 5.4 with the updated PyQT library.

Regarding the remembering the Directory Location, that seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12295

We had one of our developers take a look at how to improve this behavior and have a proposed solution here https://github.com/Ultimaker/Uranium/commit/15bb15439d7ba2d61c30a3ac2aec966a4c7cab55

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12295

closed 2023-08-28 by MariMakes