Ultimaker/Cura #12661: I am using Ender3 v2 software. At the end of where he started the model, he passes over the layer he started again and scrapes the filament of the model.

bynevzat, 2022-07-03:
### Application Version

5.0.0

### Platform

windows 10

### Printer

Ender3 pro

### Reproduction steps

1. I started track 
2.  I threw the first floor and faster than I wanted.
3. I'm back where it started, I stripped the filament

### Actual results

I canceled printing. I continue to use it efficiently with old software.

### Expected results

software review

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There may be compatibility and coordinate problems in the software.

GregValiant, 2022-07-03:
We will need a project file to troubleshoot this.
Load the model, make your Cura settings ready to slice.  Use the "File | Save Project" command to create a 3mf project file.  Zip the file.  Post the Zip file here.

bynevzat, 2022-07-03:
I am sending the project file

GregValiant ***@***.***>, 4 Tem 2022 Pzt, 00:32 tarihinde
şunu yazdı:

nallath, 2022-07-05:
Adding attachments to emails won't mean they end up with us, sorry. It's one of those things that Gihub filters out. You have to go to the web interface and upload it there.

bynevzat, 2022-10-11:
> I don't know where it went but it did not get posted here.

Didn't get the project files?  What is unwritten?

şunu yazdı: