Ultimaker/Cura #12665: Objects shift when opening 3mf file

novec001, 2022-07-04:
### Application Version

5.0.0

### Platform

Windows 10, several different computers

### Printer

Ender 3 S1 Pro

### Reproduction steps

Opened any previously saved 3mf file with objects created within Cura.

### Actual results

Main objects and parameters open just fine, but other objects are shifted -10 in the Y direction.

### Expected results

All objects should stay in their original place.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This seems to happen with any file, so no specific file is attached.

GregValiant, 2022-07-04:
I've seen this happen once or twice when I have opened other users project files but I've never been able to reproduce it myself.
So we will need a project file and your Cura.log file.  You can zip them together and post the Zip file here.

smartavionics, 2022-07-04:
Related perhaps? https://github.com/smartavionics/Cura/issues/178

novec001, 2022-07-06:
Me and my stupid assumptions... It's happened every time I've tried, so I assumed it was a common thing. My objects are added with the Calibration Shapes plugin, if that makes any difference.

In the attached project file, the cylinder is supposed to be at [-13,6,3] to control top pattern on the little boss, but ends up at [-13,-4,3]. The tube should be at [0,0,96] to remove tiny infill blobs from the threads, but ends up at [0,-10,96].

The log starts from opening a project with a single object, through creating the objects, saving as new project, closing Cura, and finally opening the project again. I started Cura by double clicking the file in Explorer, but it behaves the same way when I run Cura manually and open the project file from the menu.

[Object shift bug.zip](https://github.com/Ultimaker/Cura/files/9055740/Object.shift.bug.zip)