Ultimaker/Cura #12667: Variable/replacement patterns not being evaluated in Printer End G-code

jdulrich, 2022-07-04:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Custom FFF printer with 2 extruders

### Reproduction steps

This is the entirety of the End G-code that I set for this printer - note it includes 4 replacement patterns

```
; layer_height = {layer_height}
; machine_width = {machine_width}
; machine_depth = {machine_depth}
M104 T0 S0 ;1st extruder heater off
M104 T1 S0 ;2nd extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 ;move X to min endstops, so the head is out of the way
G90 ;absolute positioning
G1 X0 Y{machine_depth} ;Present print
;G1 X0 Y{300} ;Present print

```

### Actual results

Generated .gcode file includes the above End G-code text verbatim. The replacement patterns have not been replaced. Replacement patterns in the Start G-code are replaced correctly. There does seem to be something about it in the log.

### Expected results

I expect {layer_height} to be replaced with "0.20" and {machine_depth} to be replaced with "300", etc.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura bug.zip](https://github.com/Ultimaker/Cura/files/9039450/Cura.bug.zip)

GregValiant, 2022-07-04:
You have an error in there.  This line is illegal because "300" is not a Cura keyword.
;G1 X0 Y{300}
The result is that Cura is not doing _any_ replacements.  Change that line to:
;G1 X0 Y300
and the problem should be resolved.  Let me know how that works for you.
I'll remove the "Bug" label for now as it appears to be a syntax problem.

jdulrich, 2022-07-04:
Oops. I guess I ignored that line because it was commented out, but obviously you're doing replacements in comments. Sorry about that.

GregValiant, 2022-07-04:
You had the move covered in the previous line anyway.  I'll close this.

closed 2022-07-04 by GregValiant