Ultimaker/Cura #12668: Font size changes

Ronbo2, 2022-07-04:
### Application Version

5.0.0

### Platform

windows 10

### Printer

Ender 3

### Reproduction steps

I open Cura, either 4.18.1 or 5.0.0  and after I play a game of World of Warships or BF 2042 or Enlisted, the font in Cura has changed to a larger font, and sometimes I am unable to use the drop down menus.  

### Actual results

Here is a screensho
[image]
t

### Expected results

[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9040431/cura.log)

MariMakes, 2022-08-23:
Hey @Ronbo2,
Thanks for opening your first issue with us! 🎉 
Sorry it took us a while to reply 😢 

I think it might be related to your screen resolution/scaling. 🔍 
Do you still see this behavior if you change the scale to 100%?
[image]

Or I've also seen this behavior if I move my Cura instance to a different monitor. 
Do you use multiple monitors with different resolutions?

Ronbo2, 2022-08-23:
when I started having this problem I was running 2 FHD monitors, I now use 2 WQHD monitors.  I have a lot of other problems with Cura 5 and I am currently not using 5.x at all. Ths would occur without movint the window from one monitor to the other.

Ronbo2, 2022-08-28:
Ok, since the latest Nvidia driver update I have been unable to duplicate this problem so you can marked it as solved.  Thank you

MariMakes, 2022-08-29:
Huzzah! Happy to hear 😄