Ultimaker/Cura #12670: retraction extra prime amount, dynamic based on time of travel.

hayden-t, 2022-07-04:
### Is your feature request related to a problem?

I like the "retraction extra prime amount" setting for compensating with ooze during travel, but the amount need will vary depending on how long the travel takes and thus how much material has had chance to ooze.

### Describe the solution you'd like

If you made the amount primed per second/milli of travel time.

### Describe alternatives you've considered

post processing script

### Affected users and/or printers

ender 3 s1 pro - all metal hot end direct drive

GregValiant, 2022-07-04:
If you aren't using Z-hops you can try the "Retract Continue" post-processor.  It spreads the retraction across the following travel move.  That keeps the oozing down and can make up for the insufficient prime that occurs after long travels.
(Z-hops confuse the script and so it doesn't really work like it's supposed to if the Z is bouncing up and down).

hayden-t, 2022-07-05:
ok thanks, ill try it out !

hayden-t, 2022-07-06:
i found this too https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime

GregValiant, 2022-07-06:
I hadn't seen that one before.  It looks like exactly what you want.

idmadj, 2022-09-02:
The ScalableExtraPrime plugin works great, but it has apparently been abandoned by it's developer.

Oozing can be mitigated by lowering temps and increasing travel speed, but there are limits to those compromises. Lower temps create weaker parts and hotends can only travel so fast. 

Priming a dynamically calculated extra amount is not a perfect solution to mitigate the variable underextrusion after travels caused by oozing, as the oozed filament is still sticking on places where it shouldn't. Ideally we'd want to figure out a way to completely eliminate oozing itself, but this is the best solution we have so far. For this reason, I think it's worth considering a native implementation, as it is a pretty essential feature.

5axes, 2023-08-25:
Same type of discussion with this time a Adaptive Retraction Distance : https://github.com/Ultimaker/Cura/issues/3452

GregValiant, 2023-08-25:
@5axes I was able to get @Pheneeny 's"ScalableExtraPrime" to work by changing the API line as described in [THIS POST](https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime/issues/20) which works for 4.x versions of Cura.

I'm stuck because my laptop is in for repair so I can't check, but would there be a similar fix to get it to work with the 5.x Cura versions?

qba1990, 2023-08-25:
Thank You 5axes for pointing me that topic :),

I've just downloaded and copied to plugin destination of newest Cura 5.4.0 plugin from 5axes fork:
https://github.com/5axes/CuraPlugin-ScalableExtraPrime

And after Cura restart I can see new options for that functionality (seems to work). I'll give it a try and let You know about results.

willeywilson, 2024-07-03:
Reviving an old topic.  
I had managed to get Retract continue to work with Z hops on Cura 5.7.2.  I have downloaded and installed @5axes fork of CuraPlugin-ScalableExtraPrime.  
However, the scalable extra prime plugin only works when retract continue is not enabled.  As soon as I enable retract continue and it converts all the travel moves to G1s with reducing E values, the scalable extra prime doesn't work because in effect there are no 'travels'.  I tried this without Z Hops, with retraction and without retraction boxes ticked also.
Is it possible to have the two working together? 

i.e. retract continue modifies the code, and then scalable prime looks for any travels where the E value is reducing and considers this a travel with retract?

I have a 410x430mm bed, resulting in up to ~590mm travels from one corner to the other.  Retract continue is required to help mitigate stringing, and by the same token scalable prime is required so the first 5mm of a wall isn't missing after travel.  Particularly on PETG and the likes.

Hope this makes sense and is a simple fix.
Thanks

GregValiant, 2024-07-03:
I did a re-write of Retract Continue myself.  I just went back removed the "G0" replacement code.  That version is attached.
That might allow you to use @5axes "scalable" plugin.
[RetractContinueGV.zip](https://github.com/user-attachments/files/16091000/RetractContinueGV.zip)

willeywilson, 2024-07-05:
Thanks @GregValiant , I'll give it a go. I did actually end up installing your re-write of retract continue that allowed Z Hops, I'll see how I get on with this version.

willeywilson, 2024-07-05:
Just added the script and ran a quick test to generate a gcode and it looks like this combo works.

My code now shows a clear retract, z hop, continuous retract whilst travelling, and scalable prime.  Ignore the numbers but pasted below for anyone who comes across the same issue I did in the future.

G1 X30.76 Y36.818 E10.97971                                       - End of printed line
G1 F1800 E9.47971                                                       - Retract    
G1 F600 Z0.4                                                                 - Z hop
G0 F12000 X318.739 Y305.764 Z0.40 E5.53936 ;Trav    - retract whilst travelling as a G0
G1 F600 Z0.2                                                                 - Z unhop
G1 F1800 E11.44477 ;Adjusted e by 0.46506mm          - scalable prime
G1 F2700 X319.138 Y305.402 E11.47138                       - print next line

Thanks very much @GregValiant & @5axes

GregValiant, 2024-07-05:
I can understand the desire to maintain political correctness by changing the G0's to G1's but in a case like this where it is definitely affecting follow-on post -processors.  I don't think it's advised.
You notice I added the ";Trav" comment in an effort to appease the purists.

The original Retract Continue was stumped by z-hops and my original re-write fixed that.  I think I'll put this version in with a pull request when I get around to it.