Ultimaker/Cura #12672: Slicing error for Tough PLA

youcangetme, 2022-07-05:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

[UMS5_Bottom Shell.zip](https://github.com/Ultimaker/Cura/files/9048080/UMS5_Bottom.Shell.zip)

1. the material is set to Black Tough PLA
2. slice

### Actual results

A popup says to report a bug. I see no other data. I have sliced this very same file before but with no issues.

In fact, if you switch the material it slices just fine. So like ABS or PTEG.

### Expected results

A sliced file I can print? I can't even find a log file.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I logged for a log and could not file one. If you tell me where one is I can send it
[UMS5_Bottom Shell.zip](https://github.com/Ultimaker/Cura/files/9048142/UMS5_Bottom.Shell.zip)
.

youcangetme, 2022-07-05:
[image]
Here is a screenshot of the error and the slicing params

youcangetme, 2022-07-05:
If I spin the model around 180 then it will slice. AND! If I import the model and to not change any rotation then it will slice. It only fails when I turn the model upside down or rotate it on the red circle 180 degrees.
[image]

GregValiant, 2022-07-05:
There were a lot of bug reports on this behavior.  I've been going back over the problem models and opening them in 5.1 Beta and slicing them.  So far I have a 100% success rate slicing with 5.1 beta.
I'll close this as completed.  If you have further comments, post them here and I'll be notified.

closed 2022-07-05 by GregValiant