Ultimaker/Cura #12673: Support + Support interface Anomaly at 0.2mmm Layers

3Duke3, 2022-07-05:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 3 (but reproducible on all)

### Reproduction steps

Load file, slice as normal.

Included 2x project files for UM3, and 2x project files for Toolchanger, where issue is worse using a combination of different settings.

### Actual results

When slicing this file at 0.15mm layer height, the support structure and support interface are as expected.

However by changing the layer height to 0.2 (or using 0.2mm profile), the main support structure and interface has an unexpected outer profile shape, with the last roof interface layer having a pattern vastly different from the interface layers beneath it. 

I have tried changing many settings with various degrees of deterioation/improvement, but haven't found a combination of settings that completely resolves the issue at 0.2 layers.

Some settings make the problem worse and show similar anomalies discussed in #11646 where sometimes perimeters of shapes are traced with extruder 1 on the last interface layer. Can be seen in E3DT_0.2lay support issue.3mf project file

[image]

### Expected results

Expect support pattern and interface pattern to be the same as whats generated for 0.15mm layers, but for any layer height combination.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Support Anomalie.zip](https://github.com/Ultimaker/Cura/files/9049919/Support.Anomalie.zip)

MariMakes, 2022-08-25:
We've discussed it with the team and placed a ticket on the backlog with the intention to pick up. 
For internal reference: CURA-9564

3Duke3, 2022-08-25:
Thanks @MariMakes, for this particular case I ended up printing at 0.15 to avoid the issue. Most of the time when I see support interface issues like this, I will tweak the settings until I can see a desirable result in the preview to avoid any issue. 

So while I don't have any photos of physical prints with artifacts, I can try to explain the defects I have experienced in the past. 

With the missing "C" shaped notch to the left of the above example, when the models perimeters that should rest on the breakaway support interface above are laid down, the strands tent to droop in the notch area as they bridge across, or of the model gets further along in the print, it may start lifting/curling away from the support interface in that area which can lead to a failed print or even worse a collision that knocks the fan door open resulting in a flooded print head if not caught. This example is not so bad, but in other situations with less surface area, the perimeters may not have much to adhere too, and they can become easily unstuck and just get dragged around by the print head.

With the other odd/nonuniform concentric patterns in the support interface, this usually results in many travels and retractions as it zips around to all the different areas filling in the gaps. Sometimes the movements are too much and the nozzle will dislodge the interface layer off the underlying support structure. Other times it tends to leave some slight oozing and just a non uniform interface surface in general. This causes some sort of scaring effect in the parts supported surface, and often the very small+many deposits of oozed interface material get embedded into the parts surface, and become impossible to remove.