Ultimaker/Cura #12675: STL opening failure

KrIsChUnn, 2022-07-06:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

GEEETECH i3

### Reproduction steps

try to import a ASCII STL file generated by solidworks,

 
[image]
[failure files.zip](https://github.com/Ultimaker/Cura/files/9050832/failure.files.zip)

### Actual results

can not open STL files generated by Solidworks

### Expected results

open file successfully

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

hope you guys fix this

Asterchades, 2022-07-06:
The issue seems to be how the object is declared in the STL file. 

The file I checked leads with the line "solid Áã¼þ1" when using ANSI encoding. While this should be fine, when viewed as UTF-8 instead the space after "solid" label is missing which renders the file invalid. Changing the line to instead read "solid Test" lets the file load successfully, while converting to UTF-8 encoding also lets the file work as the space is then preserved. This leads me to suspect that Cura is attempting to load the file using UTF-8 encoding and hitting the invalid declaration.

Given the STL standard pre-dates UTF-8 by six years it would stand to reason that any STL should be read as ANSI. However UTF-8 has become the more dominant encoding standard in recent years so it isn't entirely unreasonable to expect text encountered to be in that format.

Marking this "Under Investigation" so a dev can make a final decision on whether to treat this as a Cura issue or a Solidworks one.

nallath, 2022-07-06:
You might want to re-try this for 5.1.0 beta. We did fix some things regarding STL loading and special characters.

closed 2022-07-21 by github-actions[bot]