Ultimaker/Cura #12681: [5.0.0] Start point visualization missing

printingotb, 2022-07-06:
### Application Version

5.0.0

### Platform

win10

### Printer

custom fff

### Reproduction steps

See attached project.
Enable visibility for start points in the preview.

### Actual results

Start point representation not visible on all layers/start points.

### Expected results

should be visible

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

[Seam test.3mf.zip](https://github.com/Ultimaker/Cura/files/9055682/Seam.test.3mf.zip)

Joeydelarago, 2022-07-06:
Hello, I've check out the model and I see this issue as well. The team will discuss this.

Ghostkeeper, 2022-07-13:
Seems to be a visualisation problem. The starts thing is generated in the geometry shader by looking at whether there is a travel move before the extruded move. I think it may have missed the travel move to move the Z axis because that one is not visualised in layer view.

Ghostkeeper, 2022-07-13:
Considering this issue is pretty harmless, I think we'll have to defer it.

BasF0, 2022-09-10:
> Seems to be a visualisation problem. The starts thing is generated in the geometry shader by looking at whether there is a travel move before the extruded move. I think it may have missed the travel move to move the Z axis because that one is not visualised in layer view.

I agree, I guess with the arachne engine it is now more likely that the new layer XY coordinates start exactly where the previous layer ended. The visualization can be resolved by faking that the last line of the previous layer was not an extrusion move but a travel move instead. 

I'd fix it but I'm having some trouble calculating the exact indices into the prev_line_types array to replace the type.