Ultimaker/Cura #12685: Cura 5.1.0-beta+304: Initial Layer Horizontal Expansion. Changes horizontal expansion only above the initial layer.

csterlin, 2022-07-06:
### Application Version

5.1.0-beta+304

### Platform

Windows 11, 11th Gen Intel(R) Core(TM) i7-11390H, 16.0 GB, Intel IRIS Xe

### Printer

Creality Ender 3

### Reproduction steps

Observed a grey outline outside the x- and y-axes preview display.
Asked about it online, and a reply asked if I had horizontal expansion selected, to which I replied that I have Set "Initial layer horizontal expansion" to -0.4mm.
Fiddled around with that setting and found that changing it affected not only the gray outline, but the actual edge of the object.

### Actual results

I've printed the object with "initial layer horizontal expansion" set to -0.4 in both 5.0.0 and 5.1.0 beta+304, and found discrepancy in the thickness *above* the initial layer. Specifically, in 5.0 I found the x-axis dimension to be 4.90mm. In the same location in 5.1 I found it to be 4.09mm. That difference of 0.81mm, is approximately double the -0.4mm set as the "INITIAL layer horizontal expansion. Further, I found the actual "initial layer horizontal expansion" under 5.1 to be incorrectly remained at zero, yielding the initial layer width of 4.90mm.

### Expected results

I should have found no difference in the x- and y-axes dimensions between the 5.0 and 5.1 prints in all layers above the initial layer, and, further, the initial layer x- and y-axis dimensions should also have been essentially identical.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_battery_plate.zip](https://github.com/Ultimaker/Cura/files/9057117/CE3_battery_plate.zip)
[image]
[image]

Attempted to include screen capture. Not sure if they worked.

GregValiant, 2022-07-06:
Thanks for the report.  I had no problem duplicating this in 5.1 Beta.  The setting is acting like "Horizontal Expansion" rather than "Initial Layer Horizontal Expansion".
Someone from the Cura team will take a look.

GregValiant, 2022-07-07:
Closed as a duplicate.

closed 2022-07-07 by GregValiant