Ultimaker/Cura #12686: Slicing Failed

jcwebb, 2022-07-06:
### Application Version

4.13.1

### Platform

Linux - Fedora 35 kernel 5.17.12-200

### Printer

Ender 3

### Reproduction steps

Attempted to slide a simple model.  Only unusual thing I did was with infill.  60% infill Cross 3D pattern.  It was able to slice with 60% infill and gyroid.  Went back to "Cross 3D" for the pattern and it failed again.  Slicing continued to fail even down to 5% fill.  It appears the problem is with the Cross 3D infill pattern.

### Actual results

Cura reported it could not slice.  Suggested I file bug report

### Expected results

Should have sliced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9057301/cura.log)
[Bottle_Cap_Bike_Stem.zip](https://github.com/Ultimaker/Cura/files/9057331/Bottle_Cap_Bike_Stem.zip)

GregValiant, 2022-07-07:
I opened your project file and without making any changes it sliced fine for me in 4.13.1.
Someone from the Cura team will need to take a look at the log file.

jcwebb, 2022-07-07:
Thank you.  Odd.  I was worried that perhaps I saved the project file with
the Gyroid infill that worked, but on opening the project file it does show
Cross3D and it still fails slicing.

We'll see what the log files show I guess.  Appreciate your time!

JonathanHeinz, 2022-07-07:
Same problem here, gyroid do not work for me but cross3d.

jcwebb, 2022-07-07:
Maybe the other way around Jonathan?
Gyroid DOES work for me, Cross3D does NOT.

If you're getting opposite results then clearly we are going down the wrong
path for the root cause.

Best Regards,
John Webb

Ghostkeeper, 2022-07-08:
We haven't been able to reproduce the problem yet, on our end.

The log file indeed shows more, very subtly. It shows that the engine exited with error code -6. This indicates that it got a SIGABRT - Something that could be caused by assertions failing. Assertions in CuraEngine are not compiled into our releases since we build those in release mode, not debug mode. However it could be that you found a debug build somewhere. Are you running Cura via the AppImage, or did you install it some other way?

I don't have a debug build of Cura 4.13 ready here. On the release AppImage the problem doesn't seem to reproduce, and in Cura 5.1 with a debug build also not (though there are sufficient changes since then that maybe an edge case wouldn't be hit any more).

JonathanHeinz, 2022-07-08:
My problem is in version 5.1

closed 2022-07-23 by github-actions[bot]