Ultimaker/Cura #12689: Cannot Save end g-code in Cura 5.0

oliverhbailey, 2022-07-06:
### Application Version

5.0.0

### Platform

Windows 10 Build 19044

### Printer

Creality CR10 Smart

### Reproduction steps

Open Settings -> Printer  Maintenance
Change ending Y Position from 300 to 285
Close Printer Maintenance dialog
Close Printer Dialog
Cannot slice because change was not detected or saved

### Actual results

Noting - Changes were not and cannot be saved.

### Expected results

New end g-code position should be 285, value stays at 300

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Have tried repeatedly to make a change without success.

GregValiant, 2022-07-07:
This started when the dialog boxes were re-configured for Qt6.  As a workaround, after you have made changes to either the StartUp Gcode or to the Ending Gcode, click in the opposite textbox before closing the dialog.  That should set the altered contents.
This is a duplicate of #12422 and should have been fixed in the 5.1Beta but it appears that it has not.

closed 2022-07-07 by nallath

oliverhbailey, 2022-07-07:
Actually one of the other developers sent a messsgevit is not fixed in 5.1
beta.  He provided a work around that does work.

I will not respond to your comments because you response confirms my
comments.  And my comments from 50 years of experience.  If all you have to
say was "congradulations",  you missed every point made.  Thank you.

Oliver Bailey

GregValiant, 2022-07-07:
@oliverhbailey 50+ years ago I was learning to program analog computers.  Then we got an IBM 360.  It filled a very large room with 10 or 12 cabinets.  Anyone who opened the door to the room and let the air-conditioning out was promptly flogged.  The hard discs were 36" in diameter and each had a capacity of 720k.  The punch cards were a huge improvement.

That was the end of my formal software training.  I'm a fair ME though and necessarily big on workarounds since nothing ever seems to fit as it's supposed to.  The boys and girls at Ultimaker along with the main contributors will get out their BFH's and beat this thing into shape so it's more user friendly.  I think the constant changes in the world of computing (and all the feature requests) keeps getting in the way.  I would think that having to port the program to about 5 different operating systems has to be a major headache as well.

jellespijker, 2022-07-08:
@oliverhbailey I used to be:
- paper delivery boy
- dishwasher
- cook
- silversmith
- aluminium extrusion die corrector
- aluminium extrusion die designer
- piping engineer
- basic designer engineer shipbuilding
- Assistant research and development project lead fluid dynamics
- research and development project lead fluid dynamics
- mechanical & software engineer/designer geoplymer process plants
- Mechatronic engineer Fused Deposition Modeling printhead
- Software engineer Fused Deposition Modeling slicing software

Or aren't we listing CV's?