Ultimaker/Cura #12690: Its time to overhaul the Cura Parameters screen.

oliverhbailey, 2022-07-06:
### Is your feature request related to a problem?

I have to drop down the quality list.

I have been an operating systems architect for 50 years.  Its getting very obvious that no on who works on Cura has slightest idea of how to the right hand side of the main screen.  They keep adding items, that in some cases are not even in the proper category.  Why is scaling in the material category?  Sticking it in between bed temperature and flow rate makes no logical sense at all.

### Describe the solution you'd like

Have you considered tabs, instead of an infinite list.  A little proactive design in that area would go a log way to solving problems and would force rethinking categories.

### Describe alternatives you've considered

Using another product or writing my own.  Cyra is becoming very bloated with code and it has outgrown python.

### Affected users and/or printers

Everyone, it modernizes the user interface and eliminates lists of infinite length,

### Additional information & file uploads

None.  I am not volunteering my time, I am suggesting modernization of your user interface and and moving into a compiled langaige.

nallath, 2022-07-07:
> Why is scaling in the material category?

Different materials shrink differently. As such, they require different scaling factors. To my knowledge, there isn't any other reason to change the scaling factor in the settings. 

> I have been an operating systems architect for 50 years. 

Congratulations! 

>  Its getting very obvious that no on who works on Cura has slightest idea of how to the right hand side of the main screen

I don't understand what you mean by this. What does "How to the right hand side" mean? 

> Cyra is becoming very bloated with code and it has outgrown python.

What makes you say that? I also don't see how that has much to do with the feature that you're requesting. Could you expand a bit more? It's not Python that is forcing us to use a list if that's what you're worried about. 

> Have you considered tabs, instead of an infinite list.

I has been considered in the past, but it never got a lot of priority. I do think that changes can (and at some point should) be made here, but the current view is that the visibility sets and search bar are sufficient and that other issues have more priority. 

I've looked at the QiDi interface quite recently who has a similar approach. Although it does uses tabs, it does significantly reduce the usefulness of the search bar. I'm not quite sure if that is the correct solution. Perhaps you have a good idea on how to best tackle that?

5axes, 2022-07-08:
@fieldOfView launched the developpement of a new plugin to integrate this type of interface a few week ago : https://github.com/fieldOfView/Cura-SettingsViewPlugin 
You could perhaps support him in his effort to provide users a better UI.

fieldOfView, 2022-07-08:
Please note that that plugin is far from "production ready". It will take another month or so before it is ready.

Ghostkeeper, 2022-07-08:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8121.

A bunch has been said already about this topic there, but if you have anything new to add, please add it there to consolidate the discussion.

closed 2022-07-08 by Ghostkeeper