Ultimaker/Cura #12691: Error when trying to slice model

guistela, 2022-07-07:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3 V2

### Reproduction steps

1. Tried to slice model

### Actual results

Cura couldn't slice with error message suggesting to report a bug

### Expected results

Model is not sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura_files.zip](https://github.com/Ultimaker/Cura/files/9059804/cura_files.zip)

GregValiant, 2022-07-07:
This has supposedly been fixed in 5.1Beta.
Opening your file in 5.1beta, it would not slice.
Just importing the models and using my "fine" profile it did slice.
So a conflict might be going on with your settings.  Someone from the Cura team will need to take a look.

Ghostkeeper, 2022-07-08:
Yes, this has been fixed. It slices fine now.
[image]
Thanks for the report nonetheless! It helps a lot to have people report such issues. Even if we found the issues too, they still provide a broader test for the bugfixes we make.