Ultimaker/Cura #12699: Multiple objects - one after the other - second object print starts in the air

Diegovd, 2022-07-07:
### Application Version

5.1.0 beta (also 5.0.0 official release)

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Add an object to the plate
2. Multiply it once
3. Arrange manually (since the auto arrange is a complete disaster, sorry)
4. Slice and print

Actually the travel lines give away the problem of the second object being started printed in the air, but I initially discarded that as a GUI bug.

Note: I had to increase the Gantry Height from 25 to 70mm otherwise it would not slice. Why is that a limitation if the objects are placed properly so that no collision can occurr?

### Actual results

First object prints fine. Travels to the second object location. Starts printing in the air.

First try with 2 Objects:
[image]

I was planning on doing 3 at a time, but forget about it, it will travel to the next piece and start printing in the air:
[image]

### Expected results

Object one is completed, then the second should start at the normal Z level (same as object one).

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9066488/cura.log)
[CE3PRO_PortePlaquette.zip](https://github.com/Ultimaker/Cura/files/9066496/CE3PRO_PortePlaquette.zip)

GregValiant, 2022-07-08:
When I slice your project I get this in the Gcode:
G1 F3000 E1939.04154 --------------->last retraction of the first model.
M107 ----------------------------------> fan off to start the second model
G0 F9000 X149.604 Y211.705 Z47 ----> Move the Z up
G0 X128.885 Y115.693 ----------------> Move to the area of the second model
;LAYER_COUNT:230
;LAYER:0
;MESH:PortePlaquette.stl(1)
**G0 X130.547 Y138.331 Z0.2 -----------> Drop down to the build plate**
;TYPE:WALL-INNER
G1 F3000 E1940.04154 ----------------> Prime
So the second model is starting at a Z=0.2
This is the transition to the third model and you can see that it is the same sort of thing with the model starting at Z=0.2
G1 F600 X149.384 Y119.243 E3879.98648
;TIME_ELAPSED:5654.854651
G1 F3000 E3878.98648
M107
G0 F9000 X149.384 Y119.243 Z47
G0 X127.851 Y38.233
;LAYER_COUNT:231
;LAYER:0
;MESH:PortePlaquette.stl
**G0 X129.513 Y60.871 Z0.2-----------> Drop down to the build plate**
Unless the printer misses the instruction (unlikely) then the succeeding prints would start at "Initial Layer Height" above the build plate.  If you would post that "bad" gcode file I will take a look.  (For reference - my printer is also an Ender 3 Pro.)

Be advised that if you open a "One at a time" **_gcode_** file in Cura that the preview will be wrong.  The Gcode reader will arrange the parts by layer number and so all the "LAYER:0" lines will display together.  The only accurate preview is the one you see right after slicing.

Ghostkeeper, 2022-07-08:
> Note: I had to increase the Gantry Height from 25 to 70mm otherwise it would not slice. Why is that a limitation if the objects are placed properly so that no collision can occurr?

Cura assumes more or less that the gantry is a flat ceiling. Different printers have different gantry systems of belts, rods, or arms. Cura doesn't model this all out, only the shape of the print head itself and the shape of the nozzle. So it doesn't know that you have e.g. two axis-aligned rods, one a bit higher than the other, and that placing objects on a diagonal would prevent those rods from hitting your previous object. And so Cura must assume that the gantry height + initial layer height is the limit of how high a model can be when printing one-at-a-time.

> Actually the travel lines give away the problem of the second object being started printed in the air, but I initially discarded that as a GUI bug.

After printing one object, the nozzle moves to the height of the highest print and then moves down to the initial layer height to start the print. This is necessary to prevent hitting another object during the travel move. The vertical moves are not shown in layer view as far as I know. But they should get printed.

closed 2022-07-22 by github-actions[bot]