Ultimaker/Cura #12702: Blobs everywhere with Cura 5.1.0 beta +304

Toddi3D, 2022-07-07:
### Application Version

 5.1.0 beta +304

### Platform

Windows 10 64

### Printer

Vyper

### Reproduction steps

print with Cura 5.0 all is good

print with new Cura ... Blobs everywhere on Surface

### Actual results

Bad Results with Cura 5.1.0 beta +304

### Expected results

nothing

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9067549/cura.log)

GregValiant, 2022-07-08:
A couple more items would be helpful.  With the model loaded and Cura set up to slice select the "File | Save Project" command.  Zip the 3mf file and then add the "blobby" gcode file to the Zip folder as well, then post the Zip file here.

Ergonomicmike, 2022-07-11:
I have also noticed a small "blobbing" issue after upgrading from Cura 5.0.0 to the 5.1.0 beta. A print that printed without blobs before is now blobbing. 

I suppose I should reinstall Cura 5.0.0 to make sure that the problem is following Cura.

HOLLY-DION, 2022-07-14:
Select a column, which does not automatically generate a line, but is randomized.
The Z gap problem between 5.0 and 5.1 has been bothering me for a long time.

GregValiant, 2022-07-14:
@HOLLY-DION 
@Toddi3D 
@Ergonomicmike 
We still don't have a project file.  Without knowing the settings you folks are using and the types of models your printing it's impossible to duplicate the problem.  I have been using 5.1 exclusively since about 2 hours after it's release and just haven't seen this blobbing problem and/or Z seam problem at all.

_"Select a column, which does not automatically generate a line, but is randomized."_ I honestly have no idea what that means.

closed 2022-07-28 by github-actions[bot]