Ultimaker/Cura #12704: Generator VirtualPythonEnv(file:None) failed

shaoyidi, 2022-07-08:
### Application Version

5.1

### Platform

win10

### Printer

no printer

### Reproduction steps

I followed the instruction(https://github.com/Ultimaker/Cura/pull/12544) try to install and run cura,but when I excecut the cmd "conan install . cura/5.1.0-alpha+666@ultimaker/cura_9365 --build=missing --update -o cura:devtools=True -g VirtualPythonEnv "
this problem ocured,can anyone help me?

### Actual results

Requirement already satisfied: wheel in e:\cura\cura\lib\site-packages (0.37.1)
Requirement already satisfied: setuptools in e:\cura\cura\lib\site-packages (58.1.0)
conanfile.py (cura/5.1.0-alpha+666@ultimaker/cura_9365): ERROR: Generator VirtualPythonEnv(file:None) failed
'NoneType' object has no attribute 'rootpath'
ERROR: 'NoneType' object has no attribute 'rootpath'

### Expected results

I think it will be installed successfully.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

You can tell me what files should be uploaded that helpful to solve this problem.

jellespijker, 2022-07-08:
Can you try again. I also ran into this problem and it should be fixed now by the following commits:
https://github.com/Ultimaker/conan-config/commit/c10fb89ca2910e21a6d953dc8e8d137bc6bb617c
https://github.com/Ultimaker/conan-config/commit/89300371e16650ade3686147686df5636762b1f9

yess1987, 2022-07-10:
“conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv”
The following error occurs：

protobuf/3.17.1: Appending PATH environment variable: C:\.conan\674573\1\bin
conanfile.py (cura-build-environment/None): ERROR: Generator VirtualPythonEnv(file:None) failed

ERROR:

yess1987, 2022-07-10:
did you solved？

fieldOfView, 2022-07-10:
Not all developers work in the weekend. Please practice a little more patience.

jellespijker, 2022-07-10:
I will write a short troubleshooting article on our wiki, somewhere tomorrow or the day after. For now make sure you start with a fresh environment.

```
# Start with a clean Conan slate - Windows PowerShell 
Remove-Item "$Env:USERPROFILE\.conan" -Recurse -Force
Remove-Item "C:\.conan" -Recurse -Force
python -m pip install conan --upgrade --force
conan config init
conan config install https://github.com/Ultimaker/conan-config.git
```

yess1987, 2022-07-10:
protobuf/3.17.1 package(): Packaged 1 '.exe' file: protoc.exe
protobuf/3.17.1 package(): Packaged 101 '.h' files
protobuf/3.17.1 package(): Packaged 12 '.proto' files
protobuf/3.17.1 package(): Packaged 2 '.inc' files: port_def.inc, port_undef.inc
protobuf/3.17.1 package(): Packaged 2 '.lib' files: libprotobuf.lib, libprotoc.lib
protobuf/3.17.1 package(): Packaged 3 '.cmake' files: protobuf-generate.cmake, protobuf-module.cmake, protobuf-options.cmake
protobuf/3.17.1 package(): Packaged 1 file: LICENSE
protobuf/3.17.1: Package '879e3ed2faf0bacbe180fd629dc07b37d1867ac0' created
protobuf/3.17.1: Created package revision e7c89612c9754560269c4e909fac34de
protobuf/3.17.1: Appending PATH environment variable: C:\.conan\674573\1\bin
conanfile.py (cura-build-environment/None): Generator txt created conanbuildinfo.txt
conanfile.py (cura-build-environment/None): ERROR: Generator VirtualPythonEnv(file:None) failed

ERROR:

yess1987, 2022-07-10:
I need more detailed steps, sorry

jellespijker, 2022-07-10:
I noticed the following: `cura-build-environment/None` which tells me that your're executing the steps in the wrong/old repo. PR https://github.com/Ultimaker/Cura/pull/12708 states that `cura-build-environment` will be deprecated.

Follow the instructions in:
- https://github.com/Ultimaker/Cura/wiki/Building-Cura-from-Source
- https://github.com/Ultimaker/Cura/wiki/Building-Cura-from-Source

Since the problem with the VirtualPythonEnv(file:None) has been solved with:
https://github.com/Ultimaker/conan-config/commit/c10fb89ca2910e21a6d953dc8e8d137bc6bb617c
https://github.com/Ultimaker/conan-config/commit/89300371e16650ade3686147686df5636762b1f9

I will close this issue

shaoyidi, 2022-07-11:
@jellespijker Thanks for your solution, I will try again with your instruction

closed 2022-07-13 by Joeydelarago

CHSIVASANTHOSH, 2022-12-23:
@jellespijker @nallath @rburema  i am getting this issue with 5.3.0alpha version 
conanfile.py (curaengine/5.3.0-alpha): ERROR: Generator VirtualPythonEnv(file:None) failed

I tried to remove .conan and reinstall ,removed the cache files of it ,

I wanted to run the cura form source and wanted to build it ...
Just suggest me to build cura also 🙏🙏🙏

rburema, 2022-12-23:
@CHSIVASANTHOSH The _engine_ shouldn't need a virtual python environment. Are you using the steps for the frontend to try and build the engine?

Steps to build the engine can be found here: https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source

As opposed to the front-end (building this completely will also build the engine), which is detailed here: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

CHSIVASANTHOSH, 2023-01-04:
**@rburema** @jellespijker @nallath I tried the steps https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source at this command 
> conan editable add . curaengine/latest@ultimaker/stable I am geeting stuck as  I am geeting this error **conanfile.py (curaengine/5.3.0-alpha): ERROR: Generator VirtualPythonEnv(file:None) failed**

ERROR:
from this I am unable to move further 

My main Requirement is this 
 I wanted to build .exe file with my custom name at the top of the window instead of ultimaker cura .. i customized according to my flavours ,but I am unable to change the top name , but I tried even to decode the .exe file of ultimaker cura and changed the xml   <assemblyIdentity type="win32" name=**"Ultimaker-Cura"** processorArchitecture="amd64" version="1.0.0.0"/>
but it didnt reflect still the same for this particular versions from >cura4.8 I am unable to do this 

Can you help me out in this 🙏, As I am having my own style of printers & I am 3D artist so ,I can help you out in any thing possible if you want

pitlochry, 2024-01-03:
I have this issue ("ERROR: Generator VirtualPythonEnv(file:None) failed") on cura 5.7.0-alpha.0. Versions before did not have this issue.