Ultimaker/Cura #12707: 5.1.0 beta 304 not slicing on windows 11

V1x0r, 2022-07-08:
### Application Version

5.1.0-beta+304

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

attached are the Logs, the stl file, and exported project 3mf. (added to a zip so it could be uploaded here).  

I simply opened the program, dragged the stal into Cura, and sliced with my last working configuration from 5.0.0. 

### Actual results

During the slicing procedure cura gave an error saying there was an issue with slicing and to share the log file here. 

### Expected results

Expected to Slice and provide the GCODE export for my printer.  This has worked with the prior 5.0.0 version. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Water_Tank_Villa_Pot_fixed.zip](https://github.com/Ultimaker/Cura/files/9070202/Water_Tank_Villa_Pot_fixed.zip)
[stderr.log](https://github.com/Ultimaker/Cura/files/9070215/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/9070216/stdout.log)
[cura.log](https://github.com/Ultimaker/Cura/files/9070218/cura.log)

Ghostkeeper, 2022-07-08:
I'm unable to reproduce this problem, both on the beta and on the current 5.1 branch. I'm testing on Linux though. Normally our engine bugs aren't platform-specific, but it's theoretically possible.

The log file indicates that it got a return code 3221225477 (0xC0000005) from CuraEngine, which indicates a segfault. We know of a couple of segfaults, all of which have fairly high priority. You might hit one of those cases. One of the known segfaults is also hard to fix because it's inherently a problem inside of a library we're using, but 5.0 stable might have some more fixes for you once it gets released.

I'm going to ask a colleague if they can test it on a different OS.

Thanks for the good bug template by the way! This is a nice ticket to work with.

V1x0r, 2022-07-08:
thank you for the prompt response.  ^_^  I do hope this assists in resolving the issue.  ^_^ this is good to know about linux working as expected, maybe i'll run a VM in the meantime.

GregValiant, 2022-07-08:
A little more.  This is on my Windows 10 system and Cura 5.1 beta.
I opened the file and immediately sliced it and received the "Unable to Slice" box.  That is the cura.log file in the attached zip file.
[cura.zip](https://github.com/Ultimaker/Cura/files/9074993/cura.zip)
My printer is also an Ender 3 Pro and switching to my normal "fine" settings profile and choosing to "Keep Settings" the model sliced as expected.
Opening the 3mf and switching to my printer and using the OP's "super quality" settings profile the model sliced.
Opening the STL the model sliced.
So the only problem I had was with the 3mf file provided by @V1x0r .

closed 2022-07-15 by jellespijker