Ultimaker/Cura #12710: Wall inconsistencies and behaviour

kvnper, 2022-07-09:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

custom

### Reproduction steps

slice using attached project file and cura 5.0.0 (remove '.txt' extension in filename)

### Actual results

Walls behave irregular across the same layer, and across different layers:

1. Single wall with same thickness changes from two lines to one, even having seam positions at those points, rendering the outer wall with irregularities. Pic below of layer 36 and its previous layers causing a visual inconsistency on outer wall:
[image]

2. Around the same model feature as point 1 above, the wall lines switch erratically from one line to two lines. Varying on the left and right side of the model feature, and varying by layer.

Pic of layer 12, the left side of the wall has two lines and the right side has a switch to one line midway. You can also see about 5 layers before this has the same issue:
[image]

Pic of layer 13, the behaviour has changed from the previous layers, now matching on both left and right side, different from layer 12 behaviour:
[image]

Pic of layer 29, the behaviour of the lines has changed once again, now the left side has changed to two continous lines:
[image]

Pic of layer 30, the behaviour of the line again shows a new behaviour of two continous lines on both the left and right side. This is ideal behaviour.
[image]

Finally, pic of layer 31, where the behaviour has returned to the usual behaviour here of two continous lines stopping and breaking down to one line in the middle of a straight wall.
[image]

### Expected results

Consistent behavour in line pathing. Outer walls not stopping in the middle straight line in order to draw a single thick line, causing a visual artefact on the wall.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

See above.
[IF_gripen-1.4_cura5-0-0.3mf.txt](https://github.com/Ultimaker/Cura/files/9076037/IF_gripen-1.4_cura5-0-0.3mf.txt)

kvnper, 2022-07-09:
Small update, and I should have done this beforehand, I sliced with a non-arachne cura and the model does indeed have an irregular wall it seems:

[image]

MariMakes, 2022-08-24:
Hey @kvnper,

Thanks for your report, sorry it took us a while to get back to you. 
[In Cura 5.0 we introduced a new way to build up walls and bunch of new settings. ](https://support.ultimaker.com/hc/en-us/articles/4792077687068-What-new-print-settings-are-introduced-to-Cura-5-) 

Since your object is partly hollow Cura is struggling a bit with these inner walls.
If you change your Minimum Wall Line Width to 0.3 it should display the two walls again.
[image]

Can you let us know if these new settings solve your problem? 👼

kvnper, 2022-09-07:
> Hey @kvnper,
>
> Thanks for your report, sorry it took us a while to get back to you. [In Cura 5.0 we introduced a new way to build up walls and bunch of new settings. [...]

Thanks for getting back to me. I will check out the fix as soon as I can and report back, within the coming weeks I think.