Ultimaker/Cura #12711: Outer wall deformed when doing variable line width

kvnper, 2022-07-09:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

custom

### Reproduction steps

slice using attached 3mf project file and cura 5.0.0 (Remove .txt extension from filename)

### Actual results

The outer wall is deformed when doing variable line width causing visual artefacts on the outer wall.
[image]
[image]
[image]
[image]

### Expected results

Smooth outer walls for a smooth, nice looking, quality print

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Remove .txt extension from filename

[IF_gripen-1.4_cura5-0-0.3mf.txt](https://github.com/Ultimaker/Cura/files/9076050/IF_gripen-1.4_cura5-0-0.3mf.txt)

kvnper, 2022-07-09:
Outer wall is smooth with non-arachne cura

[image]

Ghostkeeper, 2022-07-13:
This is currently expected behaviour, but I'll change this to a discussion to discuss potential avenues for how this would need to be improved.

There's a couple of requirements that create this behaviour:
* If the part is thinner than 2x the wall thickness, it should get filled completely with walls, leaving no gaps. The line widths are adjusted to make this happen.
* If there are any inner walls, the outer wall line width should never be adjusted. Only the inner wall line width may be adjusted, to reduce the visibility of line width adjustments on the outside of the print.
* If there are no inner walls, the outer wall line width will have to be adjusted, in order to meet the first requirement.

The problem occurs in that last case, where there are no inner walls. It then makes the outer wall thicker or thinner to fill the space. In theory the wall still matches exactly until the border of the object which would create a smooth wall, but in reality this doesn't happen because:
* G-code doesn't support variable line width, and so the lines have to be chopped up with a limited resolution (also to not create too detailed g-code, which is hard to keep up with for the printer's CPU).
* The centreline shifts with this variable line width. This movement creates ringing.
* The line width changes in theory, but in practice the flow rate lags behind.
* In layer view, the line joints make it render a little bit differently too.

So a remedy to reduce the problem is to make sure that there is an inner wall in these places. It'll adjust the width of that instead of the outer wall. You can do that by reducing the line width or the thresholds at which a new wall is added. This won't always solve it though, because if the part becomes smaller than 2x the outer wall line width, the outer walls are made thinner. And of course you'll then have thinner lines which are harder to extrude for the printer.

What other solutions could we think of? We would prefer not to drop the first requirement, of having no gaps there. But perhaps that is impossible then.

kvnper, 2022-07-15:
Perhaps until an elegant solution is found, implement a toggle setting where the user can decide whether the outer walls do not deform under any circumstances, even if it does leave gaps. Allow that to be used with modifier meshes and the user can specify certain critical sections only.