Ultimaker/Cura #12712: Inner wall gap fill is not working well

kvnper, 2022-07-09:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

custom

### Reproduction steps

attached 3mf project file (remove .txt extension from filename)

### Actual results

Inner wall gap fill leaves a lot of gaps on sharp corners

Layer 160:
[image]

Layer 255:
[image]

### Expected results

Gap filled with material

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

attached
[IF_gripen-1.4_cura5-0-0.3mf.txt](https://github.com/Ultimaker/Cura/files/9076065/IF_gripen-1.4_cura5-0-0.3mf.txt)

MariMakes, 2022-08-24:
Hey @kvnper,

I had a look at your file and I'm afraid it's similar to your other report https://github.com/Ultimaker/Cura/issues/12710
This behavior change was on purpose and is a bit unfortunate in the model you have. 
Sorry, it took us a while back to get back to you. 

I think you can still have your desired behavior if you change the Wall Transition Threshold Angle.
[image]

You can learn more about the new settings here:
https://support.ultimaker.com/hc/en-us/articles/360012512340

Can you let us know if these new settings solve your problem? 👼

kvnper, 2022-09-07:
> Hey @kvnper,
>
> I had a look at your file and I'm afraid it's similar to your other report #12710 This behavior change was on purpose and is a bit unfortunate in the model you have. Sorry, it took us [...]

Awesome, thank you again. Will check in the coming weeks.