Ultimaker/Cura #12713: Mouse over tooltip, hides everything under machine settings

Lupu5R3x, 2022-07-09:
### Application Version

5.x

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Trying to configuring my printer, I go to Presences -> Printers -> Choose my printer -> Machine Settings.
hover over eg. G-code flavor or trying to chose some thing from the dropdown.
Same goes for the other settings, some gives only half a window in tooltip, others full window.

### Actual results

this https://i.imgur.com/WOYxdyy.png when trying to chose G-code flavor.
Making it almost impossible to edit the settings using a mouse.

### Expected results

You should be able to change settings.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Though this isn't quit a bug, more an annoying thing - that should be easy to fix :)

fieldOfView, 2022-07-10:
The tooltip is this big because of the Settings Guide plugin. The Settings Guide plugin has an option to replace the normal Cura tooltips with the longer explanation the plugin provides. You may want to to turn that option off.

Since this is not technically Cura's fault, I will close this issue. 
Also see https://github.com/Ghostkeeper/SettingsGuide

closed 2022-07-10 by fieldOfView

Lupu5R3x, 2022-07-10:
Ahh, thanks for clarifying :) I have turned the plugin off - and now I can see what I'm doing :)

timgray, 2024-09-02:
This is still a problem in 5.8.1. and it's not settings guide as I do not have it even installed.   Please give us the ability to turn off 100% of the buit in Cura help popups.  It's impossible to edit Gcode inside the printer setup with stock clean no plugins installed cura.