Ultimaker/Cura #12715: Feature request option to zhop on combing moves.  (delta) (anything fast)

danorder, 2022-07-09:
### Application Version

5.0 / all 

### Platform

windows10/all 

### Printer

anything fast / delta 

### Reproduction steps

delta centers around hopping in most cases.  
what would be ideal.  option to enable  zhop for combing moves. notably this hinders print speed and produce excessive noise on most delta frames when the nozzle moves though infill. alternatively this tends to be more exaggerated with rubbery pla in particular with infill.  An afterthought this would also be helpful with fine layers more prone to tearing infill. eg 0.1 

### Actual results

nozzle scrapes due to not hoping here and their. disabling combing works but then print time is massive in comparison. 

### Expected results

hop and, have a note z hop will not occur on "combing moves" ugly rabbit hole for people unaware. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

.

GregValiant, 2022-07-10:
I'm not sure what you want to add/do here.  The setting is named "Z-Hop on retraction" and one of the settings in the Travel section is "Max Comb Distance with No Retraction" which would seem to cover your request?  If the Combing move was 30mm long and your setting was 10mm then there would be a Z-Hop for that move.  In addition the "Retraction Minimum Travel" setting might be involved.  Given the same 30mm combing move, if Retraction Minimum Travel was set to 35 then there would be no retraction and hence no Z-Hop even with Max Comb Distance with No Retraction set to 10.
Nozzles will occasionally scrape on infill when the infill is too high for some reason.  If your layer is being printed at 1.0mm then the top of that infill should be 1.0mm and there should be no scraping.  But I get it on some models as well.

I don't really see a bug here.  If you could more fully describe the "fix" you'd like to see I would be inclined to change this to a "Feature Request".

danorder, 2022-07-11:
Sorry this could have been worded better. I am not certain if this would be a feature request or not. Others have the same reported issue with delta. Ive had similar issues my corexy as well.   It would seem   a option to "hop" with infill in particular regardless of retraction to avoid any slight warpage.  A hop on that move from what ive seen should avoid the problem. Alternatively which could just be lack of knowledge. without combing cura appears to produce excessive retractions max retract count , max distance before retract cannot seem to fully reduce the amount of retractions relative to other slicers. Generally this produce alot of excessive hopping with infill in particular without combing.  Notably on my setup the excessive retraction typically ends up creating pressure issues or borderline disabling it just leads to strings in between short islands.  
https://github.com/Ultimaker/Cura/issues/4308
https://github.com/Ultimaker/CuraEngine/blob/main/docs/assets/retraction_combing_hop.svg
https://community.ultimaker.com/topic/20563-z-hop-during-infill/

Ghostkeeper, 2022-07-13:
If you're doing a Z hop, why would you still want to avoid hitting the walls horizontally? You're flying over the surface, so why not do a direct travel move? No need to avoid walls any more if you wouldn't hit them anyway.

But then you're basically left with the normal Z hop as it is already implemented. The current Z hop flies over everything in a straight line, without combing. However you say that this takes too much time.

One reason why it might take too much time for you, is because when it's Z hopping, it's also always retracting. This is implemented like this because the vertical movement of the Z hop necessarily also pulls out the material from the nozzle chamber, leaving a blob there if the pressure is not already dropped with a retraction. Maybe that's something to look at; doing Z hops without retraction. Is that what you're asking for? But I think you'll find experimentally that this is not a good idea.

danorder, 2022-07-14:
its not the walls in particular tops / infill usually end up having some degree of warp.  i have also noticed it will most commonly will happen with diagonal moves in particular most of the time watching it I've seen it not hop regardless of combing length.   so perhaps a always hop when moving across infill or tops regardless of retraction. would be ideal.  noticeably  its  a daily post on the flsun forum regarding scraping. but this is hard to know in particular if its user calibration issues.  The sr profiles might be worth looking.  i'll try to get some diagnostics regarding this.  another reason i can think of why a always hop option would be important would be the use of bimental breaks.

danorder, 2022-07-14:
I should add this seems to be more common with pla in particular. i cant say ive seen it involving abs or petg etc. normally i run those 2-400 without issues.

danorder, 2022-07-20:
let me gets docs or video on this. staring at it on video  9/10 it doesn't hop and cleaves though infill or wall across long paths. eg one half of the model to the other.  Im not sure what the best approach here is.  combing off its going to be constant hops or combing on intermittent scrapes. personally i find this quite excessive on infill retracting every line. ideamaker and, i think prusa slicer has a option to not retract in infill.  Notably highflow hotends seem to barely need retract. Im also not sure how much is needed in infill if at all with pressure advance or linear advance turned on. something similar may be useful for combing off.to retract less or reduce the retract / hop.  alternatively  ideamaker has the only use zhop on x feature type option. in their case only on tops , all , or only across models. personally i find only on bottoms / tops the most effective to deal with slight curling. I haven't found it to scrape with method typically it always hops one side to another on long moves without incident.