Ultimaker/Cura #12717: No Input Checking on "Top/Bottom Line Directions"

Ergonomicmike, 2022-07-10:
### Application Version

5.0.1 Beta 304

### Platform

Windows 10

### Printer

Ender 2 v3

### Reproduction steps

I typed in some angles for line directions and sometimes I forgot to enter a leading or a trailing bracket.

Example,

`0,90]`

### Actual results

If you forget a leading or trailing bracket, the top/bottom line directions do not change. (Try it with a calibration cube, first layer.)

Okay, that is expected. But there is no Warning.

### Expected results

I expected Cura to give me a Red Warning in the field, and not let me proceed to Slice until this error was corrected.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Similarly, if you type in bizarre numbers, like [720000,2], Cura "accepts" those too. Which might be okay. Except I expected to be limited to 0-360 degrees.

Ghostkeeper, 2022-07-18:
I think it's okay. I've seen at least that people sometimes use negative numbers, like `[-45, 45]`. I don't think a lot of people use 720000, but in the end it's all just getting put into an `std::sin` and `std::cos` function call in C++ and they don't care how high the value is until you hit floating point rounding errors.

Doesn't do any harm. People can fill in whatever they want here.

closed 2022-07-18 by Ghostkeeper

Ergonomicmike, 2022-07-18:
Yes, but Cura doesn't alert them if they forget a bracket.

Shouldn't it?