Ultimaker/Cura #12718: Objects printed in 5.1 beta are printing with less walls than with 5.0

GoguyT3D, 2022-07-10:
### Application Version

5.1 beta

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

While Ultimaker cura 5.0 was open on wednesday 7/6/22 I printed a prototype of a vacuum adapter I wanted. Today 7/9/22 after updating to Ultimaker Cura 5.1 beta I printed the same model while changing no settings other than filament type (From PLA to PETG) after the model printed with Cura 5.1 beta I noticed the vacuum does not fit on the model like it did with the prototype from v 5.0 I started looking much closer to the model printed with Cura 5.1 beta and noticed the initial layer was the only layer to print will all the walls and had the dimensions of what the final model should have yet the entire rest of the model was printed with I think 1 less wall thickness for the entire print. Which caused all my screw holes to be far too loose/big certian walls that needed to be 1/16" were super thin like paper and just about all walls turned out to be (let me measure) almost .8mm thinner than they should be (that being one wall on the ID and one wall on the OD missing for the whole model causing a big scaling issue. For now I will go back to 5.0 until a stable version comes out but i figured Id let you guys know.

### Actual results

Models have 1 missing wall on outer diameter and inner diameter on models with holes

### Expected results

The model should have had the same amount of walls as Cura 5.0 but it didnt. The picture shows the model standing in the orientation it was printed and the initial layer was the correct dimension of the model and the second to last layer all were incorrect dimensions to the original model from Autodesk
[image]
[image]
The blue model was printed immediately after Cura was updated from 5.0 to 5.1beta. The white model is the correct size where as the blue model was printed after the update and no settings were modified in relation to the size yet the walls were incorrect thickness and OD and ID holes were off too.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I looked very carefully at the walls and pattern used when it was printed in 5.1 beta and then uninstalled and reinstalled 5.0 stable, and noticed that a total of two walls were missing in the 5.1 beta version that were important to the model scale, one outer wall and one inner wall for a total of .8mm missing in my walls when no changes were made to the settings just a filament change

GregValiant, 2022-07-10:
Thanks for the report.  We need a project file to troubleshoot this.  Load the model in Cura and set up to slice, then use the "File | Save Project" command to create a 3mf file.  Zip the file and post it here.
One thing you can check in 5.1 is if you have "Horizontal Expansion" or "Initial Layer Horizontal Expansion" set.  It appears that they got flipped somehow and "Horizontal Expansion" is acting like "Initial Layer Horizontal Expansion" and vice-versa.  So if you have Initial Layer Horizontal Expansion set to a negative value, ALL layers are effected EXCEPT the initial layer.  That was reported in #12677 .

GoguyT3D, 2022-07-10:
Here is where it gets confusing: I always use -.35mm initial layer horizontal expansion on my models, and I never use horizontal expansion, second I have already uninstalled 5.1 so I can't make a 3mf for you, but all problems were reverted when I installed 5.0 so no biggie, I guess unless a negative initial layer horizontal expansion now means positive horizontal expansion I'm not sure what to say

GoguyT3D, 2022-07-10:
My vacuum adapter is currently printing from Cura 5.0 with the correct wall dimensions, when it is finished I will try to reinstall 5.1 beta and see if the walls change on their own again, save a 3mf and post it here. Be back in about 2 hr

fieldOfView, 2022-07-10:
Thanks in advance for putting in the extra effort to reproduce this.

GoguyT3D, 2022-07-10:
[image]
[image]
As you can see with no modification to the file or settings 5.1 beta does something to the wall count/horizontal expansion/initial layer horizontal expansion causing a 0 initial layer horizontal expansion and a positive horizontal expansion of about 1 layer per side (inside + outside) heavily effecting my need be accurate models.
[RYOBI Vacuum Adapter - Master.zip](https://github.com/Ultimaker/Cura/files/9079339/RYOBI.Vacuum.Adapter.-.Master.zip)
For safe measure I saved the Cura V5.0 installation file and like I said before I'm going to revert back to that version until a stable 5.1 comes out.

GregValiant, 2022-07-10:
_"Here is where it gets confusing"_
I would like to go on record as saying I was confused well before that.  This may just be a typo error, or maybe a "+" got changed to a "-", but "confusing" is the right word used correctly.

GoguyT3D, 2022-07-13:
Just for the record the initial layer which should have been -.35mm was
actually 0 (correct size of model) and the subsequent layers were all
smaller such that the initial layer horizontal expansion was wrongfully
applied to the horizontal expansion in the code.
Thanks,