Ultimaker/Cura #12722: "Infill line direction" setting not working ?

uski, 2022-07-10:
### Application Version

5.1.0-beta

### Platform

Linux

### Printer

Ender 3 Pro

### Reproduction steps

1. Set "Infill Line Direction" to [0]
2. Set infill to 100%
3. Set infill pattern to lines

### Actual results

Infill line direction is 45 and 135 degrees instead of 0

### Expected results

Lines should all have the same angle in all layers, 0 degrees.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Model used: https://www.printables.com/model/207243-decision-dice-parents-edition/files

GregValiant, 2022-07-10:
With the Infill at 100% Cura defaults from "Infill" to "Skin".  So you would need to change the top/bottom line directions rather than the Infill line directions.
[image]

Ergonomicmike, 2022-07-11:
Thank you for the reply.

Apparently I wasn't clear on what the Issue is.

The Issue is NOT that the numbers do not work.

The Issue is that a user can enter an invalid input - example, forgetting one or more of the brackets - but Cura will not announce the error with a red color in the field, as Cura does when users enter the wrong value in other data fields.

I'll try to change the title of this Issue if github lets me.

glx-seven, 2022-07-20:
@GregValiant 
> With the Infill at 100% Cura defaults from "Infill" to "Skin". 

This was changed in 4.13, wasn’t it? 5.1.0 (Release) on Mac also behaves as intended with the infill printing as infill. 

[image]

Ghostkeeper, 2022-07-21:
This is caused by https://github.com/Ultimaker/Cura/issues/9004 . The behaviour is intentional: When using 100% infill, it's going to print as skin as per the default settings.
We are aware that this gives confusion because the infill settings don't apply any more. Instead the skin settings apply: Top/Bottom Line Directions in this case. We don't really have a UX solution to this at the moment but we'd like to think of one. Or the problem might solve itself if we notice that many printer profiles don't use this any more.

GregValiant, 2022-07-25:
There is a workaround.  After changing the Infill Density to 100% go to the Top/Bottom section and manually change the Top Layers and Bottom layers back to whatever you had them set to.  The 100% Infill Density will not be affected but you will get the Infill Pattern that is set in the Infill section.

hoffr, 2022-07-27:
I have 100% infill and my top/bottom is using particular line direction values, but the 'infill' line directions in the gcode preview do not reflect that direction, they use the defaults as if they're `[]`. Applying those values to the infill settings also does nothing.

GregValiant, 2022-07-27:
Hello @hoffr 
If you could make up a 3mf Project File (File / Save Project) and zip it and post it here that would be good.  I don't get the issue with my setup but I might be misunderstanding what is going on.

JesperKrogPoulsen, 2023-03-24:
> We don't really have a UX solution to this at the moment but we'd like to think of one. 

You could disable the "Infill Line Directions" setting and other settings that has no effect if the infill is 100%. I get around the issue by setting infill to 99.99%