Ultimaker/Cura #12723: 5.1.0 beta installs by default over 5.0.0 Windows 10

KimmoHop, 2022-07-10:
### Application Version

5.1.0-beta+304

### Platform

Windows 10 Home 21H2, AMD Ryzen 5 4500U with Radeon Graphics

### Printer

based on Ender 3 Pro

### Reproduction steps

1. start 5.1.0 beta installer
2. blindly click OK

### Actual results

5.1.0 beta is installed over existing 5.0.0

### Expected results

5.1.0 would be installed to e.g. 5.1.0 directory by default

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Yes, it shows it's going to install to 5.0.0 directory but if it is not about uninstalling (seems to be history in 5.x?), who reads these ;)
Configuration is in 5.1.0 directory so profiles are not overwritten.

installer with default location
[image]

install location
[image]

app started
[image]

jellespijker, 2022-07-14:
Thank you for reporting this, we have this on our backlog in Jira as well.
Devs see: CURA-9459

Ghostkeeper, 2022-07-18:
Fixed in 5.1 stable!

closed 2022-07-18 by Ghostkeeper