Ultimaker/Cura #12724: ERROR: Generator VirtualPythonEnv(file:None) failed

yess1987, 2022-07-10:
### Application Version

5.0.0

### Platform

windows 10

### Printer

no printer

### Reproduction steps

1.pip install conan
2. conan profile new default --detect
3.conan config install https://github.com/ultimaker/conan-config.git
4. cd C:\cura-build-environment
5. conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv

### Actual results

Installing (downloading, building) binaries...
boost/1.78.0: Already installed!
clipper/6.4.2: Already installed!
gtest/1.8.1: Already installed!
nlopt/2.7.0: Already installed!
rapidjson/1.1.0: Already installed!
stb/20200203: Already installed!
zlib/1.2.12: Already installed!
protobuf/3.17.1: Already installed!
protobuf/3.17.1: Appending PATH environment variable: C:\.conan\674573\1\bin
conanfile.py (cura-build-environment/None): ERROR: Generator VirtualPythonEnv(file:None) failed

### Expected results

conan install success

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Need to list the specific working path and environment variable check, for novice

yess1987, 2022-07-10:
I am using the original cura-build-environment，because i don't known which conanfile.py is used?
”Execute the following command in the root directory of the Cura repository.“

fieldOfView, 2022-07-10:
Closing as a duplicate of #12704

closed 2022-07-10 by fieldOfView

yess1987, 2022-07-10:
I have updated the version but the problem is still there