Ultimaker/Cura #12725: Slicing fault with stl Object from tinkercam.com website

CGES19, 2022-07-10:
### Application Version

5.1

### Platform

Windows 10

### Printer

i3 MK2

### Reproduction steps

Load stl object from tinkercam.com website
Slice in Cura
[image]

### Actual results

The hole in the cut object is partially filled with filament
[image]
.

### Expected results

Slice Object should contain the correct hole.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I have no problem with the pruaslicer

[PI3MK2_3D-Drucker.zip](https://github.com/Ultimaker/Cura/files/9078827/PI3MK2_3D-Drucker.zip)
.

GregValiant, 2022-07-10:
You have "Make Overhangs Printable" enabled in the Experimental section.   That setting adds chamfers to overhangs so they don't need support.  Like the "Remove All Holes" setting it is useful in a very specific set of circumstances and that model isn't one of them.  In the image below I have enabled "Make Overhang Printable" for one model and turned it off on the other.
I'll close this as it is "expected behavior" with that setting enabled.

[image]

closed 2022-07-10 by GregValiant

CGES19, 2022-07-10:
> Closed #12725 <https://github.com/Ultimaker/Cura/issues/12725>  as completed.
>
>
> Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
> https://www.avast.com/antivirus

*        many thanks