Ultimaker/Cura #12726: sliceing in cura

fireelf22, 2022-07-10:
### Application Version

5.0.0

### Platform

mac

### Printer

anycubic

### Reproduction steps

didi everything normal

### Actual results

na

### Expected results

na

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

lol droid wings

fireelf22, 2022-07-10:
[image]

GregValiant, 2022-07-11:
Much of these "Slicing Failed" problems have been repaired in 5.1 beta.  Without further information we'd be unable to tell if yours is of the type that was fixed.  Workarounds are:  Rotate the model.  Change the Initial Layer Height by a little (Ex.: 0.20 to 0.21).  Move the model to a different place on the build surface in Cura.
I'll mark is as a duplicate.  If you have further information, zip a project file up and post it.

closed 2022-07-25 by github-actions[bot]