Ultimaker/Cura #12731: New Feature: "Support Retraction" - "Before" and "After"

GregValiant, 2022-07-10:
### Is your feature request related to a problem?

I don't remember this behavior prior to 5.0.  The short distance between the model and any adjacent support structure means there is no retraction when moving from the support to the model.  I am getting results where the support is welding itself to the model at the crossover between the model and the support structure.  Part of that may be the result of where the Z seam is, but it is happening on about 50% of my prints.

### Describe the solution you'd like

I propose two new settings to the Travel section.  "Support Retraction" with two sub-settings "Before" and "After".  If "Retract at layer change" were enabled then the Before setting wouldn't do much.  With Retract at Layer Change disabled "Before" would act a lot like it but only when there is support on a layer.
The "After" setting would insure a retraction when the nozzle moves from the support to the model.  In some cases the travel move  is only .6 to 1mm and that's when things get welded together.

### Describe alternatives you've considered

When moving from the support to the model across just the XY Support Distance would mean lowering the Retraction Minimum Travel setting to 0.6mm and results in too many retractions.
Enabling Retract at Layer Change on models with long combing moves results in dry runs at the start of the layer.

### Affected users and/or printers

A bunch.

### Additional information & file uploads

The circled area shows horizontal travel moves from the model to the support (and back).  This particular model suffered from both the Before move and the After move so the support was welded to the model along this line.  If there had been a retraction when moving from the model to the support at the end of a layer, and a retraction when moving from the support to the model, the heavy strings would not have occurred.
Thanks for your time.

[image]

kvnper, 2022-07-11:
I've also experienced this

GregValiant, 2022-07-13:
This is data from the gcodes of a model that I sliced several times changing just the Retract Min Travel and Max Comb Distance with no Retraction.
It reads as "Retraction Minimum Travel" **and** "Max Comb Distance with No Retract" **|** "Number of Retractions"

.2 and .2 | 3688
.4 and .4 | 3688
.6 and .6 | 3688
.8 and .8 | 3688
1 and 1  | 3688
1.2 and 1.2 | 3327
2 and 2 | 2544
3 and 3 | 2349
5 and 5 | 2158
10 and 10 | 1595
.2 and 20 | 955
20 and .2 | 662

It appears that the lower limit of "Retraction Minimum Travel" is 1.0mm.  The default Support X/Y distance for this model and printer was 0.7.  "Z overrides X/Y" was active and the Minimum X/Y distance was 0.35.
It would appear that the "Retraction Minimum Travel" cannot be made short enough to enable a retraction when moving from the support to the model.