Ultimaker/Cura #12732: Bottom Pattern Initial Layer have no effect when Spiralize Outer Contour enabled

sbasmanov, 2022-07-10:
### Application Version

5.0.0

### Platform

Mac OS 12.4

### Printer

Ender 3 Pro

### Reproduction steps

Printing thing: https://www.thingiverse.com/thing:687931 in spiral vase mode. Set Bottom Pattern Initial Layer to Zig Zag, while Top?bottom Patterns set to Concentric. Bottom Layers = 3.

### Actual results

 Cura produce all 3 layers with concentric pattern.

### Expected results

First layer with zig-zag pattern.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It may be useful to have separate settings for Top and Bottom patterns. So, N top layers will be printed with Top pattern and N bottom layers will be printed with Bottom pattern. Now there is setting for Top/Bottom (sets both), Initial Bottom Layer (only first layer) and Top Surface skin pattern (which doesn't work in spiral vase mode as well). It seems to me, existing pattern control is not straight and flexible.
And also take this settings into account in spiral vase mode, so if I set 2 bottom and 1 top layers, it will print total 3 layers with corresponding patterns.

ps: as to mentioned thing (candy bowl) I would like to have first two layers with zig-zag and third - concentric. Currently, this seems to be impossible.

GregValiant, 2022-07-11:
Thanks for the report.
Yes.  It was a change made in the Spiralize profile.  The bottom layers will be concentric only.
You are not the first to notice.  This is a duplicate of Cura Engine #1577 and Cura #12483 (which has some additional information).  I'll close this one in favor of the earlier reports.

sbasmanov, 2022-07-11:
Oh, ok. Haven't found other issues. Another question. Concentric pattern seam: should I open new issue about it? Honestly, it looks terrible :(
[image]
I printed same model with same settings in Prusa slicer and seam looks like straight line.
Thanks.

closed 2022-07-11 by GregValiant