Ultimaker/Cura #12740: interchanged entries 'Horizontal Expansion' & 'Initial Layer Horizontal Expansion'

discip, 2022-07-11:
### Application Version

5.1.0-beta+304

### Platform

Windows 10

### Printer

not related

### Reproduction steps

1. changing the **`Horizontal Expansion`** value does affect only the ***initial layer***
2. changing the ***`Initial Layer Horizontal Expansion`*** value does affect the ***whole part***

### Actual results

1. only the ***initial layer*** is *expanded* instead of the ***whole part***
2. the ***whole part*** gets *expanded* instead of the ***initial layer*** only

### Expected results

1. the ***whole part*** should be affected, if **`Horizontal Expansion`** gets changed
2. only the ***initial layer*** should be affected, if ***`Initial Layer Horizontal Expansion`*** gets changed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional information

Ergonomicmike, 2022-07-11:
When I try reversing my entries, as above, the skirt is drawn with the outer layer first, then the 3rd layer, and then the middle layer.

All zeros in both entries prints the skirt normally.

discip, 2022-07-11:
@Ergonomicmike
Apart from your observation about the skirt, does the issue described initially also apply to you?

Ergonomicmike, 2022-07-12:
Yes. I entered 10 mm for Horizontal Expansion, 0 for Initial Layer Horizontal Expansion. Only the first layer increased.

Then I entered 0 and 10, and all the layers above the initial increased in size.

Ghostkeeper, 2022-07-13:
Developers, see issue CURA-9473.

It's already fixed in a PR and will likely be merged into 5.1.