Ultimaker/Cura #12741: Z Seam alignment random does not work when combined with fuzzy skin

MADAL1EN, 2022-07-11:
### Application Version

5.1.0-beta+304

### Platform

Windows 10 64bit

### Printer

Ender3v2

### Reproduction steps

Select Z seam alignment random
Enable Fuzzy Skin

### Actual results

The Z seam seems to be non random

### Expected results

The Z Seam would appear to be random spots everywhere.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Note: This functions properly in cura 4.13.1

Ghostkeeper, 2022-07-13:
Yes, I can confirm this issue. Here is a project file that reproduces it.

[fuzzy_skin_seam_not_random.3mf.zip](https://github.com/Ultimaker/Cura/files/9102913/fuzzy_skin_seam_not_random.3mf.zip)

Ghostkeeper, 2022-07-27:
Duplicate of https://github.com/Ultimaker/Cura/issues/12190. Let's continue discussion there.

We have this issue in our planning for the next release.

closed 2022-07-27 by Ghostkeeper

nallath, 2022-09-08:
This has been fixed for 5.2