Ultimaker/Cura #12744: Slicing

Doncar06, 2022-07-12:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Neptune 2D

### Reproduction steps

Slicing baby Goku Ship

### Actual results

Didnt slice said Error

### Expected results

To slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I was asked to send a report so i did if its nothing my bad

Asterchades, 2022-07-12:
You need to include a project file for this in order for any diagnostics to be done.

At present the only assumption is that the model is in need of repair, as has been the case with _most_ such previous reports.

Doncar06, 2022-07-26:
Yea my bad I was just prompted to send a report I didn't know what I had to
sorry