Ultimaker/Cura #12745: Running cura from source PyQt6.QtNetwork module not found

phedg1, 2022-07-12:
### Application Version

CURA-9365 branch

### Platform

Windows 10

### Printer

Ultimaker Doesnt Matter

### Reproduction steps

`pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
powershell
PS cd C:\Users\Me\Documents\3D_Print\_Software\Workspace\CuraEngine
PS conan editable add . curaengine/latest@ultimaker/testing
PS cd C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura
PS conan install . cura/5.1.0-beta@ultimaker/stable --build=missing --update -o cura:devtools=True -g VirtualPythonEnv

PS C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura> .\venv\Scripts\activate.ps1
PS C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura> python cura_app.py`

### Actual results

`PS C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura> .\venv\Scripts\activate.ps1
Capturing current environment in C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura\venv\Scripts\deactivate_Activate.ps1
Configuring environment variables
PS C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura> python cura_app.py
Traceback (most recent call last):
  File "C:\Users\Me\Documents\3D_Print\_Software\Workspace\Cura\cura_app.py", line 23, in <module>
    from PyQt6.QtNetwork import QSslConfiguration, QSslSocket
ImportError: DLL load failed while importing QtNetwork: The specified module could not be found.`

### Expected results

I expected this to run cura from source.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm trying to run Cura and CuraEngine from source. I am on the latest commit in the Cura-9365 branch for Cura and the latest commit in the 5.1 branch in CuraEngine. I ran the commands as listed in the wiki page for running from source. I have tried restarting the powershell and rebooting my pc. No dice. What should I be doing for this?

yess1987, 2022-07-14:
I have this problem too, how to fix it

SWDontheway, 2022-08-16:
> 
i also have this issue and i'm now trying your intermittment solution.
what all ".dll" in which folder do you  mean?
i've copied all ".dll" file from the `C:\Cura\venv\Lib\site-packages` to `C:\Cura\venv\Lib\` or 'C:\Cura\venv\Scripts'...
but both didnt work.

ZacharyHowell-Essentium, 2022-08-23:
@jellespijker 
I'm also still having this issue. Has there been any solid solution formed yet?
Thanks!

jellespijker, 2022-08-23:
Working on this as we speak, I believe the problem is that PyQt6 and Qt6 link to the system python3.dll and not our own python310.dll.

I'm currently setting things up to compile Qt6 and PyQt6 from source in our build process, which is not a trivial thing, but I'm making progress.

I will reopen this issue since it is still an ongoing issue

ZacharyHowell-Essentium, 2022-08-23:
@jellespijker 
Great, thanks for the update!
Was the bot supposed to auto-close this issue?

jellespijker, 2022-08-23:
No the bot wasn't suppose to close this issue....
**Bad bot**

jellespijker, 2022-08-23:
#12773 and #13047 are both duplicates of this btw

closed 2022-08-23 by jellespijker

SWDontheway, 2022-08-29:
@jellespijker thanks for updating!

How do I use this commit to solve the issue? Could you pls provide more infos. 

I retried the build commands again from https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source. But still the same error. 

the python venv was activated, showing python 3.10.4.

jellespijker, 2022-08-29:
See the discussion on #12773 still work in progress atm