Ultimaker/Cura #12747: Cura Crashes on Startup

jacauc, 2022-07-12:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

N/A

### Reproduction steps

Launch Cura

### Actual results

Cura window becomes unresponsive and have to kill it from task manager

### Expected results

Cura launches properly. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9091291/cura.log)

Ghostkeeper, 2022-07-13:
Can I ask a bit about your computer monitor set-up?

* Do you have multiple monitors? And if so...
* Do you have multiple graphics cards? And if so, are they assigned to render each a different monitor?
* Do your monitors have different display scales? E.g. one set to 200% scale because they are high resolution or something.

I'm asking this to see if this could be the same issue as https://github.com/Ultimaker/Cura/issues/11929.

jacauc, 2022-07-13:
Hi, yeah I have three monitors all connected to one graphics card, all monitors are exactly same model and on the same scale. Two connected with displayport and the 3rd on HDMI.

Is there a known solution to the problem to prevent it from recurring?

trivalik, 2022-07-14:
see https://github.com/Ultimaker/Cura/issues/11929 , disable secondary/third monitor was a workaround.

jacauc, 2022-07-14:
> see #11929 , disable secondary/third monitor was a workaround.

This is not a suitable workaround in my configuration as it's hard to shut down the monitors in the way it is installed, but I'll consider this in the meantime.

Hoping for a more permanent fix though

jacauc, 2022-07-14:
Will give it a try thanks. Was looking for a fix instead of a workaround though. This should help in the meantime