Ultimaker/Cura #12751: Color in Flow to show Retractions

Ergonomicmike, 2022-07-13:
### Is your feature request related to a problem?

Stringing.

### Describe the solution you'd like

I would like to be able to see when Retractions are scheduled so that the user can tweak Retraction Minimum Travel, plus Retraction speeds.

### Describe alternatives you've considered

I looked at Flow and Speed to see if I could see when Cura is commanding retraction. For example, I have my Prime speed set lower than Retract, so I expected to see a change in colors when resuming flow. I've zoomed in a lot. If Cura shows the speed or flow changes, I'm not seeing it.

### Affected users and/or printers

All

Ghostkeeper, 2022-07-13:
It's not shown in the Flow and Speed colour schemes. After all, those need to show the flow and speed. The flow will be 0 or negative for travel moves (it can be negative if Retract-Continue is used in a .gcode file). The speed should be the travel speed.

However it is shown in the Line Type colour scheme. Light blue moves are retracted. Deep blue moves are not. You can see the difference well in this screenshot, where some external travel moves are not retracted due to a (laughably) low maximum retraction count:
[image]

closed 2022-07-13 by Ghostkeeper

Ergonomicmike, 2022-07-18:
Thx. The light blue lines showing retraction weren't "intuitively obvious" to me, especially since there's already a dark blue color. (American spelling.) Perhaps change the retraction color to orange? Or Brown?

BTW, the reason that some users might set their Maximum Retraction Count to a laughably low setting is because a few "expert" websites are suggesting that the default 100 is too high (causes filament failure) and that 10 is a better number.