Ultimaker/Cura #12755: Constant failure to slice

Salted-Denim, 2022-07-13:
### Application Version

5.1.0-Beta+304

### Platform

Windows 11 x64

### Printer

Anycubic I3 Mega S

### Reproduction steps

Literally anything I try to slice will either slice or not - if it doesn't slice, it gets to the same spot in the loading bar, pauses for a nanosecond, the curser blinks and then the error message that "Slicing failed with an unexpected error. Consider reporting a bug on our issue tracker." This happens with STLs that I have sliced with no issue in the past and with new ones. I can sometimes resize the thing to 99% (or any other random amount) and it will then slice, sometimes repositioning it on the buildplate gets it to slice. I've even noticed that there was one time that if I made beyond "X" number of edits in Cura, it would start throwing that error message out there. But then there are times, like now, that I have just turned on the computer, just freshly opened Cura and it instantly cannot slice any STL. This is beyond ridiculous and is happening to other folks in my local 3D community. Are there any fixes or workarounds? I don't have octoprint, so that's an option for me. Thanks for any and all help! 

### Actual results

Slicing fails, nonstop

### Expected results

The program to actually work as intended.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

Salted-Denim, 2022-07-13:
I don't know how to get logs or anything like that to try to help any further, otherwise I would've added those onto here. Sorry.

Asterchades, 2022-07-14:
There were details in the template when you first posted explaining how to acquire a log and project file. They will be necessary for any diagnostics to be done in this instance.

---

- For slicing issues, upload a project file that clearly shows the bug.
To save a project file go to `File -> Save project`. Please make sure to .zip your project file. For big files you may need to use WeTransfer or similar file sharing sites.
G-code files are not project files!

- A log file for crashes and similar issues.
You can find your log file here:
Windows: `%APPDATA%\cura\<Cura version>\cura.log` or usually `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
MacOS: `$USER/Library/Application Support/cura/<Cura version>/cura.log`
Ubuntu/Linux: `$USER/.local/share/cura/<Cura version>/cura.log`

  If the Cura user interface still starts, you can also reach this directory from the application menu in Help -> Show settings folder

closed 2022-07-28 by github-actions[bot]