Ultimaker/Cura #12757: Not slicing correctly

vic014, 2022-07-13:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

CR10s

### Reproduction steps

Opened file in Cura, went to preview, this is what i see.
[image]

### Actual results

Not sliced properly 

### Expected results

Sliced object ready for printing 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing else

Asterchades, 2022-07-14:
Without a log or project I can only guess, but it would appear that you've either told it not to display the wall lines (the settings would be just above where the provided image stops at the top) or you've told it that the wall width and skin thickness are 0mm.

I would hazard a guess that the "Shell" line type has simply been disabled in this case. But you will need to provide more information should that not be the case.

Ghostkeeper, 2022-07-18:
Yeah, the pictures are identical. We'll close this. It was a user error, having accidentally disabled the visualisation of the shell.

vic014, 2022-10-11:
> I saw that same image on Reddit<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.reddit.com%2Fr%2FCura%2Fcomments%2Fvye5zg%2Fthis_a_glitch_in_the_matrix_or_a_wrong_setting%2F&data=05%7C01%7C%7C989b92cb7b044e63209e08da6597f43f%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637934000566747422%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=8Z6Wj8pOWaH66rtHuWDYoU7362ukD721RmzBy7zguLM%3D&reserved=0> and it appears that this was sorted out by enabling the Shell linetype? Can this be closed?

Yes it can

Thank You

Victor Lawver